nodemcu/nodemcu-firmware #2029: Keyboard Emulation Support

nilsbrychzy, 2017-07-05:
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature
HID.h so https://github.com/arduino-libraries/Keyboard could be used

### Justification
The NodeMCU has a mcro usb plug so you can connect to pc and complete automated tasks.

### Workarounds
- Remote Keyboard Control (like remote shutdown)
- Smart Home (like "Hey Alexa, start netflix")

8<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2017-07-05:
From what I understand this is essentially a dup of #1917.

ccxx72, 2017-07-05:
I think it's the opposite of #1917.
In the #1917 they need to connect some HID periferals to the board and use the periferal as input.
In this case we need to use NodeMcu, or others devboard with an USB port, as a HID keyboard. 
Shortly, it's a keyboard emulation. I don't know if it's possible. Teensy can do that.

pjsg, 2017-07-05:
It isn't possible on Nodemcu as the USB chip is a dedicated serial port USB chip. There is a software USB implementation: http://hackaday.com/2016/09/03/software-usb-on-the-esp8266/ but I have no information about it. It *might* be able to do what you want -- but it certainly isn't integrated into the lua firmware.

nilsbrychzy, 2017-07-10:
Thanks! I'm new in that stuff so yeah

closed 2017-07-10 by nilsbrychzy