nodemcu/nodemcu-firmware #2030: spiffsimg creates incompatible filesystem for small fs sizes

jmattsson, 2017-07-06:
### Expected behavior
The bundled `spiffsimg` to generate usable filesystems.

### Actual behavior
For small file systems (128k or less I believe), NodeMCU does not recognise the filesystem and promptly reformats it.

### Test code

In `user_config.h`, something like:
```
#define SPIFFS_FIXED_LOCATION 0xA0000
#define SPIFFS_FIXED_SIZE 0x10000
```
Prepare a filesystem of that size:
```
tools/spiffsimg/spiffsimg -f test.fs -c 0x10000 -i
> import Makefile Makefile
```
Build and flash firmware + fs at the appropriate spots. Boot and you'll get told that it's formatting the filesystem, and your Makefile is nowhere to be seen.

### NodeMCU version
`dev`

### Cause
In the olden days we determined the logical block size to use depending on the overall filesystem size. This logic remains in `spiffsimg/main.c`. In the NodeMCU `spiffs.c` however, the handling of this has changed to be incompatible with `spiffsimg`. I don't see an obvious way of resolving this conflict, as now the address depends on the block size, and the block size depends on the size, which in turn depends on the address. @pjsg I'm very tempted to lob this into your court to figure out :)

pjsg, 2017-07-06:
Yeah -- I'll take a look tonight.....

pjsg, 2017-07-07:
I've built what may be a fix in https://github.com/pjsg/nodemcu-firmware/tree/spiffs-blocksize which I forked off from dev. This builds, and it *ought* to work, but I haven't tested it....

PR #2038 (2017-07-12): Fix the SPIFFS blocksize issues
Fixes #2030.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] @jmattsson has thoroughly tested my contribution.

This change makes the mounting code try and mount the filesystem with both the small and the large blocksize. Thus it shou [... cut]
closed 2017-07-12 by marcelstoer