nodemcu/nodemcu-firmware #2031: ENDUSER_SETUP with custom html page

ShAzmoodeh, 2017-07-06:
hi 
I want to use customized html page of original html page in enduser module but its not work !
I use docker for compile and rebuild 

I do it :

1-in "eus" folder I changed some string in enduser_setup.html to other language and save it, .gz file is make with 7-zip and replace with original file - compile with no error and flash it . but module run original web server !

2-for resolve it I try change destenition folder in enduser_setup.lua from "enduser_setup.html" to "eus/enduser_setup.html" and recompile it but again its run original web server 

3-i canceled to change original string to other language and just changed to English language with other character and again recompile and flash it but again .....

4-i change name of file from enduser_setup.html to myhtml.htm and renamed it on enduser_setup.lua , but again original web server is running ....

i do this ways and cant to resolve it , even in every cycle I erased flash with esptool.py then reflash it

I think in every change , enduser_setup.lua run in error with this file and run backup html web server .. 

  

### NodeMCU version
Master branch , 2.1.0 version

### Hardware
ESP-07

ShAzmoodeh, 2017-07-08:
??? can help me for run alternative html page???

marcelstoer, 2017-07-08:
See #1010. I just uploaded my custom HTML page with [NodeMCU-Tool](https://github.com/AndiDittrich/NodeMCU-Tool) - still works.

closed 2017-07-08 by marcelstoer