nodemcu/nodemcu-firmware #2037: Add remaining time to tmr.state()

ziotibia81, 2017-07-11:
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature
May be useful a function to get the remaining time of the timer.

### Justification
At this time I can start a timer with a desired interval, but there is no way to know how much time is missing at expiration.

8<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2017-07-15:
If I'm not mistaken this may not be very reliable as the timer action is not _guaranteed_ to run exactly in the defined interval. Depending on what else is running on the device task execution may be delayed.

ziotibia81, 2017-07-16:
Tu have a Not so reliable remaining time is better than not have it at all...
In many application could be enough. An option could be the time elapsed since start. When tmr.ALARM_AUTO is used this could be unknown without use extra code.

marcelstoer, 2017-07-16:
In light of #1010, my previous comment and @dnc40085 's thumbs up to that I'm closing this. Fellow committers, feel free to reopen if you disagree.

closed 2017-07-16 by marcelstoer