nodemcu/nodemcu-firmware #2040: WS2812 module timing problems

Skyp3, 2017-07-12:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

Leds are colored according to the pattern given in "data"

### Actual behavior

Already the first led has an incorrect color:

ws2812.write({pin=4,data=string.char(1,0,0)}) <- First led is blue, brightness 5
ws2812.write({pin=4,data=string.char(2,0,0)}) <- First led is blue, brightness 6
ws2812.write({pin=4,data=string.char(4,0,0)}) <- First led is blue, brightness 7
ws2812.write({pin=4,data=string.char(8,0,0)}) <- First led is blue, brightness 8
ws2812.write({pin=4,data=string.char(16,0,0)}) <- First led is red, brightness 5
ws2812.write({pin=4,data=string.char(32,0,0)}) <- First led is red, brightness 6
ws2812.write({pin=4,data=string.char(64,0,0)}) <- First led is red, brightness 7
ws2812.write({pin=4,data=string.char(128,0,0)}) <- First led is red, brightness 8

This works also in combination:
ws2812.write({pin=4,data=string.char(8+128, 0, 0)}) gives purple (red + blue)

ws2812.write({pin=4,data=string.char(0,1,0)}) <- Second led is green, brightness 5
ws2812.write({pin=4,data=string.char(0,2,0)}) <- Second led is green, brightness 6
ws2812.write({pin=4,data=string.char(0,4,0)}) <- Second led is green, brightness 7
ws2812.write({pin=4,data=string.char(0,8,0)}) <- Second led is green, brightness 8
ws2812.write({pin=4,data=string.char(0,16,0)}) <- First led is blue, brightness 1
ws2812.write({pin=4,data=string.char(0,32,0)}) <- First led is blue, brightness 2
ws2812.write({pin=4,data=string.char(0,64,0)}) <- First led is blue, brightness 3
ws2812.write({pin=4,data=string.char(0,128,0)}) <- First led is blue, brightness 4

# The next four always work only the second time when ws2812.write is called
ws2812.write({pin=4,data=string.char(0,0,1)}) <- First led is green, brightness 5
ws2812.write({pin=4,data=string.char(0,0,2)}) <- First led is green, brightness 6
ws2812.write({pin=4,data=string.char(0,0,4)}) <- First led is green, brightness 7
ws2812.write({pin=4,data=string.char(0,0,8)}) <- First led is green, brightness 8

ws2812.write({pin=4,data=string.char(0,0,16)}) <- Second led is green, brightness 1
ws2812.write({pin=4,data=string.char(0,0,32)}) <- Second led is green, brightness 2
ws2812.write({pin=4,data=string.char(0,0,64)}) <- Second led is green, brightness 3
ws2812.write({pin=4,data=string.char(0,0,128)}) <- Second led is green, brightness 4

ws2812.write({pin=4,data=string.char(128,0,0)}) <- First led is red (should be green, strip is GRB)

### Test code

See above

### NodeMCU version
dev-esp32 latest

It works flawlessly with an esp8266 and the corresponding nodemcu-branch

### Hardware

esp32-devkit C

8<------------------------ END BUG REPORT -------------------------------------

devsaurus, 2017-07-12:
You supply the led strips with 5V and have a level shifter between the ESP and the strip input?
I've seen such symptoms when connecting the ESP output directly to a strip with 5V supply.

Skyp3, 2017-07-12:
@devsaurus Yes, the led strip is at 5v, the ESP is powered by micro-usb. The signal input is directly connected to pin 4, ground is common to both.

devsaurus, 2017-07-12:
> Yes, the led strip is at 5v, the ESP is powered by micro-usb. The signal input is directly connected to pin 4

The first LED can't reliably detect the 3V3 output level of the ESP in this configuration. That's pure electronics, there's nothing we can do in the firmware.
You either need to add a level shifter or power the WS2812 from 3V3 instead of 5V. Adafruit's excellent guide addresses this in detail: https://learn.adafruit.com/adafruit-neopixel-uberguide/power#driving-5v-neopixels-from-3-dot-3v-microcontrollers

Skyp3, 2017-07-14:
Thanks for your comments, I will dive into this and close this issue in the meantime.

closed 2017-07-14 by Skyp3