nodemcu/nodemcu-firmware #2041: File writing (or reading?) sometimes fails.

Aterocana, 2017-07-13:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
I open a file in writing mode, I write some data in it and after closing it I'm able to read the data I wrote.
### Actual behavior
Sometimes, after writing data on the file, reading it shows no result (like the file is empty), even if the following chunk says it has been created and has the correct size:
```Lua
l = file.list();
for k,v in pairs(l) do
  print("name:"..k..", size:"..v)
end
```

### Test code

```Lua
f = file.open("data", "a+")
f:write("hello\n")
f:close()
print(file.open("data"):read())
```
I also tried to force the write operation with a flush (`f:flush()`).
Then I open the file in read mode:
```Lua
f = file.open("data", "r")
print(f:read())
```
and sometimes I get empty result (sometimes it works, with the same snippet).
### NodeMCU version
2.1 master branch

### Hardware
Many nodemcu boards
8<------------------------ END BUG REPORT -------------------------------------

Aterocana, 2017-07-31:
I've noticed it happens often when I write the same file very often (once each second). Any news?

TerryE, 2017-07-31:
>  I've noticed it happens often when I write the same file very often (once each second). Any news?

IMO, you need to read up on the life-cycle limitations of nand flash memory. Even with SPIFFS carrying out its own form of wear levelling, this is a good way to trash your ESP module and get bizarre errors. If you need high frequency write then keep the data in RAM or log to a remote node using WiFi.

This isn't really a bug in NodeMCU as you'd get exactly the same sort of problems if you called SPIFFS with this usage pattern from your own C application coded directly over the SDK.

Aterocana, 2017-08-01:
Ok, thanks for your suggestion. I'd close this issue then.

closed 2017-08-01 by Aterocana