nodemcu/nodemcu-firmware #2043: srv:close() want free memory...

chathurangawijetunge, 2017-07-14:
The srv:close() function works it shut down the server but it does not free the memory.
Is it a bug?

Aterocana, 2017-07-14:
Are you talking of a regular TCP server? I noticed that too.

chathurangawijetunge, 2017-07-14:
Yes regular TCP server...

marcelstoer, 2017-07-14:
@chathurangawijetunge / @Aterocana What branch or firmware version? Do you have a reproducible test case?

chathurangawijetunge, 2017-07-15:
Firmware- custom build (cloud)
Branch- dev

I'm trying to set up a TCP server on port 80 and get some user data.
After getting the data I want to close the server on port 80 and open a nother server on a different port.
Srv: close() does the job...
But it want free the memory acquired by the previous server
Is it a bug?

Aterocana, 2017-07-15:
Monday I'll prepare a reproducible test on branch master. Thanks.

TerryE, 2017-07-16:
The usual issue here is programmer error, and not a bug; e.g. you are closing the upvalue rather than the local copy.  This leaves a dangling reference in the registry so the `__gc` method isn't fired.

chathurangawijetunge, 2017-07-16:
So then how to close a TCP server property....

chathurangawijetunge, 2017-07-16:
So then how to close a TCP server property....
Clearing all local copies....

TerryE, 2017-07-16:
Sorry but this is standard Lua, so please read the FAQ and see the documentation [example 3](https://nodemcu.readthedocs.io/en/master/en/modules/net/#example_3).  Note that the `net.socket` close uses `sck` not `conn`or  `sv` and thememory won't be fully released until you stop the server as well.

closed 2017-07-16 by TerryE

chathurangawijetunge, 2017-07-17:
Still same issue...

marcelstoer, 2017-07-17:
> The usual issue here is programmer error

That was my initial reaction as well. @chathurangawijetunge that's why we're insisting on self-contained test cases (i.e. Lua code) in our issue template.

chathurangawijetunge, 2017-07-17:
Found a way.... by calling

Conn:on("recive",nill)
Conn:on("sent",nill)
Before
Conn:close()