nodemcu/nodemcu-firmware #2044: Wifi stops working after receiving 32 UDP packets to open port

Skyp3, 2017-07-14:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

Wifi should continue to work :)

### Actual behavior

Wifi stops working after receiving 32 UDP-Packets to an open UDP socket. The size of the packets or the delay between the packets does not matter, also other packets (ICMP echo requests for example) do not count. The following messages on serial console can be observed some seconds after wifi stops working:

I (31576) wifi: bcn_timout,ap_probe_send_start
I (34076) wifi: ap_probe_send over, resett wifi status to disassoc
I (34076) wifi: state: run -> init (1)
I (34076) wifi: pm stop, total sleep time: 0/22846853

I (34076) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:

If the UDP socket is not open (or the packets are sent to another port, which does not listen), this behaviour does not occur.

### Test code
```Lua
wifi.mode(wifi.STATION)
wifi.start()
wifi.sta.config({ssid="myssid",pwd="mypassword"})
net.createUDPSocket(net.UDP, 20):listen(1234)
```

Then send packets with:
for I in $(seq 1 32); do echo 1 | nc -q0 -u $NODEMCUIP 1234; sleep .1; done;

### NodeMCU version
dev-esp32

### Hardware
esp32 devkit C

8<------------------------ END BUG REPORT -------------------------------------

devsaurus, 2017-07-16:
> I (31576) wifi: bcn_timout,ap_probe_send_start

This is another example of the consequences when violating the multithreading requirements of lwip, as @jmattsson [discovered recently](https://github.com/nodemcu/nodemcu-firmware/issues/2014#issuecomment-311635579). Shares same root cause with #1970 and #2014.

Please have a look at https://github.com/devsaurus/nodemcu-firmware/tree/esp32-netconn. I've ported the UDP part in the meantime and the issue you described is not reproducible there.

Skyp3, 2017-07-18:
@devsaurus I can confirm that the code seems to work, thanks for your effort. How to proceed with it, can someone authorized merge it with the dev-esp32 branch?

devsaurus, 2017-07-18:
Thanks a lot for checking!
I'm currently polishing the code and will start a PR soon.

jmattsson, 2017-07-22:
This has now been merged. Is it ok to close this one?

closed 2017-07-25 by marcelstoer