nodemcu/nodemcu-firmware #2049: Trouble connecting NodeMCU to Microsoft Azure IoT Hub

georeb, 2017-07-19:
I am trying to connect my ESP8266, running the 1.5.4.1 NodeMCU build, to a Microsoft Azure IoT Hub via MQTT Protocol.

It appears that this is possible, as it is shown here...

http://thinglabs.io/workshop/esp8266/sending-d2c-messages/

I am using the correct syntax as far as I can see from the MS Azure help...

https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-mqtt-support

Unlike the example in ThingLabs, which creates a SAS token using SNTP, I have followed the MS document and generated a SAS token using the Device Explorer, for testing purposes.
**THIS SAS TOKEN WILL EXPIRE AT 23:59 ON 26/07/17**

My LUA code with **LIVE Azure IOT Hub** is as follows...

```
-- Create variables

DEVICE = "testdevice"
IOTHUB = "mynewiothub.azure-devices.net"
PORT = 8883
USER = IOTHUB.."/"..DEVICE.."/api-version=2016-11-14"
PASS = "SharedAccessSignature sr=mynewiothub.azure-devices.net%2Fdevices%2Ftestdevice&sig=i02twAZnoWS%2B%2BJk%2B65kCSx%2FQ9aAlOFivBYh7%2FeS%2FHps%3D&se=1501110735"

-- Create an MQTT Client

azure = mqtt.Client(DEVICE, 60, USER, PASS)

-- Connect to IoTHub via MQTT

azure:connect(IOTHUB, PORT, 1, 0, 
    -- Callback for a successful connection
    function(client)
        print("Connected")     
    end,
    -- Error callback, if connection fails
    function(client, reason)
        print("Error Connecting: "..reason)
    end
)
```
The response is always "Error Connecting: -5" which translates to "There is no broker listening at the specified IP Address and Port".

I suspected that NodeMCU was struggling with the SSL (port 8883) connection. So I ensured that I was able to make HTTPS connections (after verifying the SSL root certificate) with the following code, this worked but did not help.

```
http.get("https://example.com/info", nil,
function (code, resp)
     print(code, resp)
end)
```
However, I've noticed that running this after a fresh boot...

`net.cert.verify(true)`

and then re-trying the http.get command, I get the following response...

```
HTTP client: Disconnected with error: -28
HTTP client: Connection timeout
```

I cannot seem to get this to work, any thoughts?!

Perhaps someone could run the above code on 1.5.4.1 branch and report back??

# **Willing to make PayPal donations for people's time.**

Many thanks in advance.

### NodeMCU version
1.5.4.1 Frozen (working SSL)

### Hardware
ESP8266-12F

marcelstoer, 2017-07-20:
2 SO questions for the record:
- https://stackoverflow.com/q/41834637/131929 (yours)
- https://stackoverflow.com/q/40602293/131929

georeb, 2017-07-26:
Really hoping someone can test this for me please?!

The recent jump to version 2.x has rendered this firmware completely useless for my particular application.
The SSL issue apparently should not affect version 1.5.4.1, however i'm discovering that this may not be the case with MQTT at least!

Also the `net.cert.verify(true)` (above) issue is not documented...?

Could someone please support me with this?
Again, happy to donate for your time.

Thanks in advance.

nwf, 2017-08-03:
@georeb If you are able to produce custom builds, I would be very curious to see the debug output of nwf/nodemcu-firmware@5be80ab put through its paces for your use case.  You will probably want to read the commit logs to see what I've done, and might need to tweak some of the build options.

There's a great deal of subtlety involved in SSL certificate verification and even more heap space.  I fear you will not be able to make it work with RSA certs for keys bigger than 1024 bits or so, but I could be mistaken.  ECC keys seem more likely to work (and I have gotten them to work, locally, admittedly with some pain and anguish).

You might wish to read what's been observed in #1707.

georeb, 2017-08-10:
Thanks for your response, but I'm sorry @nwf, I am not able to produce custom builds, YET! Soon.
Can ECC keys be used with my application - Azure??

nwf, 2017-08-11:
@georeb I tried connecting to `mynewiothub.azure-devices.net:8883` using `openssl s_client -showcerts -connect mynewiothub.azure-devices.net:8883`.  The result indicates that `ECDHE-RSA-AES128-SHA256` is negotiated, as the key is a 2048-bit RSA key, which makes me think it wasn't terribly likely... but I gave it a shot anyway, and am glad I did.

Starting from 38784 bytes of heap free, the low water mark during a connection attempt with certificate verification on was 10096 bytes free (so about 28Kbytes used), and the connection was successful at the TLS layer.  It did, however, yield an error at the MQTT level: "MQTT: CONNACK REFUSED (CODE: 5)".  I am not sure what to make of that.  Without certificate verification, the low watermark was 13536 bytes free, and again, MQTT raised the error above.  In both cases, after the connection was established and MQTT packets were flowing, it looks like the heap had about 22K free.  You may experience slightly higher numbers (i.e., less heap usage) due to the fact that I am using a debug build for testing.

This suggests that, at least with the latest mbedtls integrated in tree, you can, indeed, connect to Azure's IoT hub.  You will almost surely have to connect early in initialization and will be likely unable to reconnect without restarting, unless your application is extremely minimal, as the connection attempt uses a great deal of heap.  However, if you can live with that constraint, you should be good to go, TLS-wise.

georeb, 2017-08-23:
Thankyou very much @nwf for your help with this issue, it is very much appreciated! :)
Your testing has given me hope of getting this working!!

The reason you are getting "MQTT: CONNACK REFUSED (CODE: 5)" was due to an expired SAS token. I have updated the SAS token (please see above) and would be very grateful if you could run the test again and see if you can fully connect??

After a lot of messing around (hence the late reply!) I have gotten one step closer...

It appears to be a certificate issue my end. I still don't get to an MQTT stage, but fall down on SSL authentication.

I have tried 3 different scenarios...

1. No certificates loaded to flash ---> ERROR -5 (as before)
2. Single StartCom certificate loaded (as per [this documentation](https://nodemcu.readthedocs.io/en/1.5.4.1-final/en/modules/net/#netcertverify)) ---> ERROR -5 (as before)
3. Loading 4 certificates, as explained in [this topic](https://nodemcu.readthedocs.io/en/1.5.4.1-final/en/modules/net/#netcertverify) ---> ERROR client handshake failed

It was delighted to see that NodeMCU was now talking to the Microsoft server, as you can see below...
```Lua

enter mqtt_socket_connect.
TCP ip is set: 255.255.255.255
TCP port is set: 8883.
leave mqtt_socket_connect.
> enter socket_dns_found.
TCP ip is set: 40.118.27.192
enter socket_connect.
leave socket_connect, heap = 37312.
leave socket_dns_found.
certificate 662
client handshake start.
distinguished names: [Microsoft IT SSL SHA2], [*.azure-devices.net]
before 1663266944, tv_sec 1663267212, after 1694370944
distinguished names: [Microsoft IT SSL SHA2], [Microsoft IT SSL SHA2]
distinguished names: [Microsoft Corporation], [Microsoft Corporation]
distinguished names: [Microsoft IT], [Microsoft IT]
check the signature ok
distinguished names: [Baltimore CyberTrust Root], [Microsoft IT SSL SHA2]
before 1556130944, tv_sec 1556131213, after 1682361344
look for a trusted cert
distinguished names: [Baltimore CyberTrust Root], [StartCom Certification Authority G3]
client handshake failed
Error: No trusted cert is available
enter mqtt_socket_reconnected.
enter mqtt_socket_disconnected.
leave mqtt_socket_disconnected.
leave mqtt_socket_reconnected.
```
So the question I have now is, which certificates am I missing?!
My knowledge of certs is limited but I'm trying my hardest to understand them. I did the research and found [this page](https://www.startcomca.com/root) from StartCom which has a vast amount of certs to download.

I was hoping, given the new info above, that you would be able to steer me in the right direction please?

Many thanks in advance :)

nwf, 2017-08-23:
@georeb I believe, but have not tested to confirm, that I used the leaf certificate as dumped by `openssl s_client -showcerts -connect mynewiothub.azure-devices.net:8883`.  It should, in principle, work with either.

Those certificates, specifically, are for the subjects `/CN=*.azure-devices.net` (leaf) or `/C=US/ST=Washington/L=Redmond/O=Microsoft Corporation/OU=Microsoft IT/CN=Microsoft IT SSL SHA2` (CA).

Unless you're speaking to other services over SSL, you shouldn't need any other certificates loaded.

nwf, 2017-08-24:
@georeb I just now ran your code on my test device and was rewarded with "MQTT: Connected".

georeb, 2017-08-24:
Thanks @nwf I shall take a look and report back.

Good news that you were able to connect. Was this test performed on an actual ESP device running 1.5.4.1? Also did the available heap improve after the successful connection?

nwf, 2017-08-24:
Erm, no, this was on an actual ESP running my dev-ssl branch.

The heap hit about 10.5K free during SSL bringup and about 22K free after connection establishment.

georeb, 2017-08-24:
Okay @nwf I have found and loaded the certificates you mentioned, thank-you.
Although it still refuses to connect to the broker, it appears that the certificates are no longer an issue as you can see below...
```Lua
enter mqtt_socket_connect.
TCP ip is set: 255.255.255.255
TCP port is set: 8883.
leave mqtt_socket_connect.
> enter socket_dns_found.
TCP ip is set: 40.118.27.192
enter socket_connect.
leave socket_connect, heap = 39936.
leave socket_dns_found.
certificate 37b
client handshake start.
distinguished names: [Microsoft IT SSL SHA2], [*.azure-devices.net]
before 1663266944, tv_sec 1663266976, after 1694370944
distinguished names: [Microsoft IT SSL SHA2], [Microsoft IT SSL SHA2]
distinguished names: [Microsoft Corporation], [Microsoft Corporation]
distinguished names: [Microsoft IT], [Microsoft IT]
check the signature ok
distinguished names: [Baltimore CyberTrust Root], [Microsoft IT SSL SHA2]
before 1556130944, tv_sec 1556130976, after 1682361344
look for a trusted cert
distinguished names: [Baltimore CyberTrust Root], [Baltimore CyberTrust Root]
distinguished names: [Baltimore], [Baltimore]
distinguished names: [CyberTrust], [CyberTrust]
use the CA certificate for signature verification
check the signature ok
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 5.
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 4.
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 3.
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 2.
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 1.
event timeout. 
Can not connect to broker.
Error Connecting: -5
keep_alive_tick: 0
leave mqtt_socket_timer.
```
However, you've really thrown a curve ball with your custom build! I was hoping that your testing was done on 1.5.4.1, as mentioned. What are the differences between the 'frozen' branch and your 'dev-ssl' branch? Is it a available for testing here please?

Correct me if I am wrong, but I don't believe that I am suffering #2042 as I am not getting any SNTP messages.

Many thanks for your ongoing support :)

nwf, 2017-08-24:
My dev-ssl branch is based on dev, which includes an entirely new SSL stack, and my branch in particular incorporates updates courtesy of @djphoenix.  It's available where you might expect: https://github.com/nwf/nodemcu-firmware/commits/dev-ssl ; you'll have to build your own images.  It's not ready for prime time, but at least I have no interest in attempting to debug anything more about the 1.5.4.1 (old) SSL stuff, since I think its utility is strictly decreasing with time.

No, indeed, #2042 does not seem relevant.

georeb, 2017-09-21:
@nwf I have been playing with your custom dev-ssl branch and have successfully connected to my Azure hub. I am delighted, thank-you. However...

I am unable to publish payloads to Azure. The MQTT connection breaks down and disconnects. Are you aware of this? Is there a solution for this issue?

Unfortunately, as you mentioned earlier, it requires an enormous amount of RAM and therefore I have very little remaining memory to play with, until #2068 is resolved I guess...? :(

nwf, 2017-09-21:
@georeb I am unaware of the issue and have no real advice to offer, I'm afraid.  It may be interesting to get packet traces and TLS debugging messages from the ESP.  (You should even be able to get the ESP to give up its session keys for decoding the packet traces, if you're sufficiently motivated.)

georeb, 2017-09-22:
@nwf I have unlimited motivation for this project but unfortunately, that is slightly beyond my capabilities.

However, hopefully the output below may shed some light...??

```Lua
esp8266:publish("testtopic","123",0,0, function(client) print("Data published successfully.") end)
enter mqtt_socket_publish.
Sent: 16
TLS<2> (heap=17776): ssl_tls.c:6864 => write
TLS<2> (heap=17776): ssl_tls.c:2701 => write record
TLS<2> (heap=17776): ssl_tls.c:1258 => encrypt buf
TLS<2> (heap=17776): ssl_tls.c:1560 <= encrypt buf
TLS<2> (heap=17776): ssl_tls.c:2416 => flush output
TLS<2> (heap=17776): ssl_tls.c:2435 message length: 85, out_left: 85
TLS<2> (heap=16184): ssl_tls.c:2441 ssl->f_send() returned 85 (-0xffffffab)
TLS<2> (heap=16184): ssl_tls.c:2460 <= flush output
TLS<2> (heap=16184): ssl_tls.c:2850 <= write record
TLS<2> (heap=16184): ssl_tls.c:6892 <= write
send_if_poss, queue size: 1
publish, queue size: 1
leave mqtt_socket_publish.
> enter mqtt_socket_timer.
timer, queue size: 1
event_timeout: 5.
enter mqtt_socket_sent.
sent1, queue size: 1
Data published successfully. --APPEARS TO HAVE SENT SUCCESSFULLY!!?
send_if_poss, queue size: 0
sent2, queue size: 0
leave mqtt_socket_sent.
TLS<2> (heap=18192): ssl_tls.c:6907 => write close notify
TLS<2> (heap=18192): ssl_tls.c:4032 => send alert message
TLS<2> (heap=18192): ssl_tls.c:2701 => write record
TLS<2> (heap=18192): ssl_tls.c:1258 => encrypt buf
TLS<2> (heap=18192): ssl_tls.c:1560 <= encrypt buf
TLS<2> (heap=18192): ssl_tls.c:2416 => flush output
TLS<2> (heap=18192): ssl_tls.c:2435 message length: 69, out_left: 69
TLS<2> (heap=16600): ssl_tls.c:2441 ssl->f_send() returned 69 (-0xffffffbb)
TLS<2> (heap=16600): ssl_tls.c:2460 <= flush output
TLS<2> (heap=16600): ssl_tls.c:2850 <= write record
TLS<2> (heap=16600): ssl_tls.c:4045 <= send alert message
TLS<2> (heap=16600): ssl_tls.c:6923 <= write close notify
Reason:[-0x7880]
enter mqtt_socket_timer.
timer, queue size: 0
keep_alive_tick: 1
leave mqtt_socket_timer.
TLS<2> (heap=19640): ssl_tls.c:7065 => free
TLS<2> (heap=29840): ssl_tls.c:7130 <= free
enter mqtt_socket_reconnected.
Error Connecting: -5
enter mqtt_socket_disconnected.
MQTT Disconnected.
PANIC: unprotected error in call to Lua API (stdin:4: attempt to call global 'connect' (a nil value))
```

Another question is, if SSL worked correctly with 1.5.4.1 and people are reverting back to that version in order to connect securely, then why won't MQTT work using it?

I'd like to use the 1.5.4.1 master branch because it is the last known **working** and **stable** version.

Although you say that it is an old version that you're not willing to support anymore?
Surely if it is the last working SSL version, then it is worth supporting?

nwf, 2017-09-22:
@georeb So far as I know, none of us are paid to work on nodemcu; I, certainly, am not.  So no, it's not "worth supporting," at least to me, because I don't use 1.5.4.1 and have no need to do so.  If you wish to discuss financial arrangements for support, I'm sure there are people who would be receptive.

I cannot answer your question about MQTT not working on 1.5.4.1; I would bet, however, that there are bugfixes to MQTT in master that are not there in 1.5.4.1.  I believe my patch for handling zero-length payloads is in such a state, for example.

Your debugging log suggests (since 0x7880 is MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY) that Azure is hanging up on you for some reason.  I am unable to advise further.

marcelstoer, 2018-07-20:
This has stalled with no progress for long -> closing

closed 2018-07-20 by marcelstoer