nodemcu/nodemcu-firmware #2050: dht_read() reads data in a wrong way?

LoveDaisy, 2017-07-20:
At line 122, 123 and 182, 183 in file [dht.c](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/dhtlib/dht.c), a macro is used to combine lower and higher bits of sensor data bytes. The macro is defined as
```c
#define COMBINE_HIGH_AND_LOW_BYTE(byte_high, byte_low)  (((byte_high) << 8) | (byte_low))
```
I think it is equivalent to byte_high * 256 + byte_low, while it should be byte_high * 10 + byte_low according to documents of dht sensor (say [this one](https://akizukidenshi.com/download/ds/aosong/DHT11.pdf)). 

The code works fine with dht11 sensor because all lower bits are 0 and the app won't come to this line. I'm afraid it will go wrong when reading other dht sensor (say dht12).

I might misunderstand the document or the code, but it confused me. Can any one tell me am I making mistakes? Is there something wrong?

AdrianDroid, 2017-07-23:
DHT 12 user here.

the reading is all WRONG!

i have  500+ degree C , adn Humidity 1000+

LoveDaisy, 2017-07-24:
@AdrianDroid Have you tried changing the macro from

```c
#define COMBINE_HIGH_AND_LOW_BYTE(byte_high, byte_low)  (((byte_high) << 8) | (byte_low))
```

to

```c
#define COMBINE_HIGH_AND_LOW_BYTE(byte_high, byte_low)  (((byte_high) * 10) + (byte_low))
```

which I think might solve your problem.

AdrianDroid, 2017-07-24:
No, but it should work but this will break other DHT like DHT 22
i think it is better off  to create dht_read12() function  or  fix dht_read11() checksum, as DHT 11 and DHT 12 is compatible models

marcelstoer, 2018-02-24:
No feedback from the author on the PRs -> closing this as well

closed 2018-02-24 by marcelstoer