nodemcu/nodemcu-firmware #2058: ENDUSER SETUP

ShAzmoodeh, 2017-08-02:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
connect to router and pull ip from router dhcp under all channel wifi

### Actual behavior
if channel number of wifi router isnt equality with channel of esp8266 in standalone mode for connect,  crashed and run in ap_mode with "setup_gadget" name ,..
username and pasword of wifi router correctly stored in flash before crash and when restart esp8266 it can connect automatically to wifi router if auto connect mode is enabled,
when esp8266 crashed , in web server page stuck on TRYING ... step!

i tried with same channel number in wifi router and esp8266 and evry thing is ok and esp8266 have ip 

i understood it when actived debug mode in "enduser_setup.c" and its show this masage :

874:         enduser_setup_serve_status_as_json
701:  	enduser_setup_http_serve_header
373:  	deferred_close
353:  	handle_remote_close
1276: 	enduser_setup_http_connectcb
243:  	enduser_setup_check_station
298: 	       AP_CHAN: 1, STA_CHAN: 3
308: 	       Turning off Station due to different channel than AP
1359: 	enduser_setup_ap_start
1388: 	SSID: ICEL_023F7A, CHAN: 1
227:  	enduser_setup_check_station_stop

### Test code

this code run in init.lua

eustmr=tmr.create()
eustmr:register(10000,tmr.ALARM_SEMI,function()
enduser_setup.stop()
eustmr:unregister()
end)
wifi.setmode(wifi.STATIONAP)
        local wificfg={}
        wificfg.ssid="test"
        wificfg.pwd="1234567890"
        wificfg.auth=wifi.WPA_WPA2_PSK
        wifi.ap.config(wificfg)
        enduser_setup.manual(true)
        enduser_setup.start(function() print("success") eustmr:start() end,print,print)

### NodeMCU version
MASTER 2.1.0

### Hardware
ESP-07 with 1MB flash

8<------------------------ END BUG REPORT -------------------------------------

@marcelstoer

ShAzmoodeh, 2017-08-05:
i just comment this lines Temporarily 

      **if (currChan != state->softAPchannel) {
        state->lastStationStatus = curr_status;
      
        ENDUSER_SETUP_DEBUG("Turning off Station due to different channel than AP1");
      
        wifi_station_disconnect();
        wifi_set_opmode(SOFTAP_MODE);
        enduser_setup_ap_start();
      }**

in 272 to 284 lines and

**else
  {
    ENDUSER_SETUP_DEBUG("Turning off Station due to different channel than AP2");
    wifi_station_disconnect();
    wifi_set_opmode(SOFTAP_MODE);
    enduser_setup_ap_start();
  }**

in 306 to 312 lines in "enduser_setup.c" and solve this problem and esp can stable to connect with router with valid ip .. but "onconnect" function in first argument of enduser_setup.start() function cant run when esp optained vali ip . so i set timer in lua file to alarm in every secound and when esp optained valid ip that teurned off enduser_setup with calling "enduser_setup.stop()"

but its Temporarily :)

Ampakinetic, 2017-08-30:
Thank you for posting your solution, this problem affected me today at a client's premises and I looked like an idiot because I'd tested and retested the solution with my router but it wouldn't work at their office.

Ampakinetic, 2017-09-05:
I thought I'd expand on what my issue was in case that provides additional information.

The main problem the original reporter of this bug and myself were experiencing is that "pick the strongest AP channel and stick to it" doesn't work well if the environment is noisy with many APs.  

When I was testing in my lab the AP is right next to me and it works perfectly every time.  However my client's premises is a busy industrial area so there are about 20 APs in the list and the one that they wanted to connect to is not the strongest one, and is on a different channel.  

Also I was using EUS manual mode, so the automatic mode "SetupGadget_XX" is not ideal and the connect event never fires so it's essentially a show stopper in this situation.  

The units I've developed are meant for fire alarms and so when they originally get setup by the technician they will be in many unpredictable places and so even knowing about the strongest signal issue will not help when deploying the units as they are wired into a panel.  The proposed solution above seems reasonable, because it stops the EUS from crashing on the wrong channel, and continues as expected when an IP address is obtained from DHCP.

I can see how checking the credentials and reporting success can be a problem if the ESP needs to stay on the same channel to confirm that the connection succeeded.  

Maybe the visual confirmation could be optional as on my side there are indicators and a reset button to clear incorrect credentials.  This way the unit could stop being an AP sooner, and try to connect on any channel before firing the success event?

ShAzmoodeh, 2017-09-05:
@Ampakinetic you tested my solution ? i not sure its work in noisy area . but now its work in my place . perhaps near 10 ssid i see in my device area but it solution work fine .

marcelstoer, 2017-09-05:
@ShAzmoodeh and @Ampakinetic if you're sufficiently confident that your changes have no negative side effects for other use  cases we'd of course be happy about a PR to that effect. Then those who've [been working on this module in the past](https://github.com/nodemcu/nodemcu-firmware/commits/dev/app/modules/enduser_setup.c), namely @jfollas @FrankX0 and @jmattsson, could give their opinion.

ShAzmoodeh, 2017-09-16:
i'm sure this is a bug . but this solution is temprarily and not a good solution for this bug

linuxslate, 2017-10-29:
End User Set up has not worked for me since a 2016 build.  I can't use any of the features of newer builds if EUS is non-functional.

My Code:

`
--[[WiFi Connection Check - Called directly from init.lua
Passes control to clock itself]]--

if wifi.sta.status() ~= 5 then  --Bad WiFi

    file.remove("static.lua") -- New Network, delete old settings
    uart.write(0,"\r   CONFIGURE WIFI   ")
    --print("CONFIGURE WIFI")
    wifi.setmode(wifi.STATIONAP)
    wifi.ap.config({ssid="VFD Clock Setup", auth=wifi.OPEN})
    enduser_setup.manual(true)
    enduser_setup.start(
    function()
    --print("Connected to wifi as:" .. wifi.sta.getip())
                  --12345678901234567890
        ssid = string.sub(wifi.sta.getconfig(),1,14)
        uart.write(0,"\rSSID: "..ssid) 
        --enduser_setup.stop()
        --wifi.setmode(wifi.STATION)
        ssid = nil
	dofile("vfd_clock.lua")
    end,
    function(err, str)
    --print("enduser_setup: Err #" .. err .. ": " .. str)
    	uart.write(0,"\r WIFI SETUP ERROR   ")
    end
)

else -- Good WiFi

	--enduser_setup.stop()
	wifi.setmode(wifi.STATION)

	--check for static IP settings file.
	if file.exists("static.lua") then
 	  dofile("static.lua")
	end
	uart.write(0,"\r IP "..wifi.sta.getip())
  	dofile("vfd_clock.lua")

end
` 

When I connect to the 'VFD Clock Setup' AP, It correctly searches for networks.

It sees 2 APs in the building on the same network as 2 separate networks, with no indication of signal strength (I am right next to one, and the other is on another floor, so much weaker).

No matter which I pick, it gets stuck on TRYING ... step as reported above.

If I reload the web page, it then says "No Networks Found (204)"  (The "204 error code is from memory, but the point is that is will no longer find any AP's

Resetting and trying again produces same result.  (and I tried many times -- it never connected).  It never shows up in the router's client lists.

limbo666, 2018-01-29:
Well this was a real nightmare. I had to try again and again until I found this post.
I can confirm that this bug exists. 
Setting my home WiFi channel to the same as ESP (a simple _print(wifi.getchannel())_ helped me on the case) was the solution but there should be a fix on firmware. End user setup is very useful option.

jmd13391, 2018-01-29:
Unless I'm missing something here, the ESP8266 has a single radio. With my limited understanding of ESP's internals, I would think that would require both the ESP's internal AP and the external STA AP to be on the same WiFi channel. If my thinking is correct, then when using the ESP's STA+AP mode, if the external STA AP changes WiFi channel (ex: when it's in "Auto" channel mode, rebooted, etc.), client connections to the ESP's internal AP will fail (because they are now on a different channel).

It seems to me that some sort of logic is needed to keep the WiFi channels in-sync when using STA+AP mode (unless of course, that logic already exists). ;-)

-Joe D.

devsaurus, 2018-01-30:
> the ESP8266 has a single radio

Right, this statement is backed by the explanations in https://bbs.espressif.com/viewtopic.php?t=324. The linked article is quite old, but its content is repeated in Appendix A of current NON-OS SDK API Reference manual.

Though I'm not sure how this fact relates to the initial problem description.

marcelstoer, 2019-07-21:
This module has recently seen a major overhaul in #2810. Maybe the affected users can try again with a build from the `dev` branch (until the PR lands on `master`) and report back if this is still buggy.

HHHartmann, 2019-07-27:
I tested it but without `enduser_setup.manual(true)` so there is a chance that it still does not work in this scenario.

closed 2019-08-10 by stale[bot]