nodemcu/nodemcu-firmware #2061: tmr memory leakish?

nwf, 2017-08-04:
### Expected behavior

While playing about with dynamic timers, I witnessed what looked like a memory leak.  When a tmr.create()'d object goes out of scope, even if the timer is pending, I expect it to remove itself from memory after firing and for the heap to go back to the state it was in before.

### Actual behavior

Instead, some trace lingers.  It looks like perhaps a dynamic array is grown up to some limit and never shrunk?

### Test code
```Lua
> wifi.setmode(wifi.NULLMODE)
> nodde.restart()
...
> print(node.heap())
44344
> for i=1,100 do tmr.create():alarm(i*100,tmr.ALARM_SINGLE,function() print(i, node.heap()) end) end
...
97	41328
98	41464
99	41600
100	41736
> i = nil
> collectgarbage()
> print(node.heap())
41752
```

All told, there are 2592 heap bytes unaccounted for.  Doing that for-loop
again drops the heap to 41216 bytes (536 more bytes unaccounted for).  Subsequent
iterations do not further consume the heap, but the 3128 bytes allocated are
seemingly gone forever, thus raising the possibility that some dynamic structure is grown
to some limit and never freed.  I'd not worry about it, save that that's a decent percentage
of the total heap space!

Conceivably, whatever's happening is happening inside the espressif firmware; I'm not sure.
If this proves to be unfixable, it should perhaps be noted in the documentation.

### NodeMCU version

dev, specifically e09e830d4b7d00265577292dc620e445641c7184 with some should-be-irrelevant changes.

### Hardware

Generic 32Mb flash ESP board.

TerryE, 2017-08-04:
It's not a leak. Your description is correct, but this is how Lua works.  Read up on the Lua registry in PiL and dump out `debug.getregistry()` to understand why.

nwf, 2017-08-04:
Oh, I see, the registry is forever the largest it ever was.  Ick!  I would have expected embedded lua to make a different tradeoff there.  Sorry for the noise.

closed 2017-08-04 by nwf

TerryE, 2017-08-05:
@nwf Nathaniel, I did think of adding a `node` function to allow applications to trim back the Registry, but in the end it's not really worth it as it only grows too large in pathological conditions.