nodemcu/nodemcu-firmware #2065: ws2812.write() malfunctions when called from timer callback

Skyp3, 2017-08-06:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
ws2812.write() should set the leds correctly, even when called from a timer callback

### Actual behavior
Leds sometimes get wrong color information and subsequently show a wrong color when called from a timer callback. When called from the main lua thread, it works as expected

### Test code

Does not work:
```Lua
t = tmr.create()
t:register(70, tmr.ALARM_AUTO, function () ws2812.write(pin=14, data=string.char(255,0,0):rep(256)) end)
t:start()
```

Works:
```Lua
while true do 
    ws2812.write(pin=14, data=string.char(255,0,0):rep(256))
end
```

### NodeMCU version
esp32-dev

### Hardware
esp32 devkit c

devsaurus, 2017-08-12:
Your test code works for me (braces `{` are missing) on a strip with 8 LEDs. Same for the chaser from http://nodemcu.readthedocs.io/en/dev-esp32/en/modules/ws2812/#example_1
There seems to be another dependency.

Do things improve if you reduce to `:rep(8)` temporarily?

Skyp3, 2017-08-16:
@devsaurus yes, it doesn't occur with 8 leds only. With 256 leds it mostly seems to occur further down in the chain, not at the very beginning. But since it only flashes for the fraction of a second, it is quite hard to tell exactly.

Skyp3, 2017-08-16:
@devsaurus Is it possible that it runs on another core when called from the timing callback and the ws2812b-routine is preempted by something with higher priority sometimes? (i.e. WLAN or sth like that)

devsaurus, 2017-08-17:
When the timer expires, it'll start its own callback but this redirects immediately to the nodemcu task. Thus I don't expect any differences in this direction although I can't explain why you observe the hiccups only when running from a timer.

If something like preemption or even interrupt masking happens for >30µs then the ISR would get into trouble to refill the RMT buffer with new patterns.

devsaurus, 2019-05-31:
Closing due to inactivity.

closed 2019-05-31 by devsaurus