nodemcu/nodemcu-firmware #2067: SNTP FEATURE REQUEST

ShAzmoodeh, 2017-08-07:
8<------------------------ FEATURE REQUEST ------------------------------------

### Missing feature
ADD decimal number to sntp.setoffset() like 1.5 , 2.5 , 3.5 , 4.5 , .... or -1.5 , -2.5 , -3.5 , ....

### Justification
many region time have this offset's

8<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2017-08-08:
Seems to me that you're (ab)using the SNTP offset for timezone handling? It's intended for a different use case, see https://nodemcu.readthedocs.io/en/latest/en/modules/sntp/#sntpsetoffset and https://stackoverflow.com/questions/44166414/nodemcu-time-zone-based-sntp-synchronization.

closed 2017-08-08 by marcelstoer

pjsg, 2017-08-09:
If you want to do timezone handling, then take a look at https://github.com/nodemcu/nodemcu-firmware/tree/master/lua_examples/timezone which is a complete implementation of timezone support.

ShAzmoodeh, 2017-08-23:
@pjsg can help me to try it step by step??
i try this module with this way:
1-upload eastern.zone in file system with nodemcu-tool command
2-upload tz.lua module in file system
and write this example code
 ```
tz = require('tz')
 tz.setzone('eastern')
 sntp.sync(nil, function(now)
   local tm = rtctime.epoch2cal(now + tz.getoffset(now))
   print(string.format("%04d/%02d/%02d %02d:%02d:%02d", tm["year"], tm["mon"], tm["day"], tm["hour"],         tm["min"], tm["sec"]))
 end)

```
and this returned 👍 
 tz.lua:32: attempt to index global 'struct' (a nil value)

marcelstoer, 2017-08-23:
You need the NodeMCU `struct` module in the firmware.

ShAzmoodeh, 2017-08-23:
ow yeah ,, its solved .. now how its seem work correctly . 
now for example how i try to set timezone . here i upload eastern.zone in file system ... how to chenge offset? because its offset sets to -5 and seem its not make for change it

ShAzmoodeh, 2017-08-23:
i founded it from that source 

> On my linux system, these files can be found in /usr/share/zoneinfo.

thanks alot