nodemcu/nodemcu-firmware #2069: ESP32: Netconn receive issue

larsstenberg, 2017-08-10:
### Expected behavior
Since netconn was introduced i have problems when receiving data:
First packet from server:
0x89 0x01 0x00 0x88 0x01 0x00 0x8a 0x78 
Second packet from server
0x89 0x01 0x10 0x88 0x01 0x02 0x8a 0x1E

In the receive method i receive 0x89 0x01 0x00 0x88 0x01 0x00 0x8a 0x78  twice. What the server sends is verified with wireshark.

### NodeMCU version
dev-esp32

devsaurus, 2017-08-10:
Thanks for reporting.

I would need some example Lua code and a simple server functionality (maybe with netcat) that reproduce and investigate this issue.

devsaurus, 2019-05-31:
Feel free to re-open and provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.

closed 2019-05-31 by devsaurus