nodemcu/nodemcu-firmware #2072: MQTT not reaching the broker

netomarin, 2017-08-13:
### Connecting to the broker and be able to send / receive messages

### The module is not connecting (I've already tested with other devices and the MQTT broker is visible)

### Test code
m=mqtt.Client(node.chipid(), 120)
m:connect("192.168.1.43", 1883, 1, 0, function(client) print("connected") end, function(client, reason) print("failed:"..reason) end)

### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: master
	commit: b62fae918b3627209fe434950cffb2506cee9a31
	SSL: true
	modules: bit,cron,crypto,encoder,file,gpio,http,i2c,mqtt,net,node,ow,perf,pwm,sjson,spi,struct,tmr,uart,websocket,wifi,tls
 build 	built on: 2017-08-13 05:05
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)

### Debug
enter mqtt_socket_client.
NodeMCU_a38932
length username: 0
length password: 0
MQTT: Init info: 10717490, , 
leave mqtt_socket_client.
> 
m:connect("192.168.1.43", 1883, 1, 0, function(client) print("connected") end, function(client, reason) print("failed:"..reason) end)
m:connect("192.168.1.43", 1883, 1, 0, function(client) print("connected") end, function(client, reason) print("failed:"..reason) end)
enter mqtt_socket_connect.
TCP ip is set: 192.168.1.43
TCP port is set: 1883.
enter socket_connect.
leave socket_connect, heap = 38088.
leave mqtt_socket_connect.
> client handshake start.
please start sntp first !
client handshake failed!
Reason:[-0x7880]
enter mqtt_socket_reconnected.
failed:-5
enter mqtt_socket_disconnected.
leave mqtt_socket_disconnected.
leave mqtt_socket_reconnected.

### Hardware
ESP 07

djphoenix, 2017-08-13:
> `m:connect("192.168.1.43", 1883, 1, ...)`
I see you have set `secure` flag for server with port `1883` which usually used to plain (non-TLS) connections. Maybe there is a reason? Are you server TLS-ready?

netomarin, 2017-08-14:
I've used both (0 and 1) and had the same error.
Ps.: No, it's a local development server running in my machine, non-TLS.
Thanks

aderbas, 2017-08-21:
For me, module already connected, listening to server topics, receiving data and responding but the server says: "Error 406", the module response was not accepted? Do not understand.

[image]

crapthings, 2017-10-31:
i'm getting failed -5 too esp12e

dtran123, 2017-11-19:
Me too I am getting -5 error. 

here is my lua code:

local mqttclientid = "theduke"
local mqtttopic = "duketest"
local mqttkeepalive = "120"
local mqttuser = ""
local mqttpassword = ""
local mqttcleansession = "1"
local mqttlwtopic = "dead"
local mqttlwmesssage = "iamdead"
local mqttlwqos = "0"
local mqttlwretain = ""
local mqttserver = "iot.eclipse.org"
local mqttserverport = "8883"
local mqttsecure = "1"
local mqttsubscribeqos = "0"
local mqttpublishqos = "0"
local themessage = "wow super"
local mqttpublishretain = "1"
local m = mqtt.Client(mqttclientid, mqttkeepalive, mqttuser, mqttpassword, mqttcleansession)
m:lwt(mqttlwtopic, mqttlwmesssage, mqttlwqos, mqttlwretain)
m:connect(mqttserver, mqttserverport, mqttsecure, 1, function() m:subscribe(mqtttopic, mqttsubscribeqos, function() m:publish(mqtttopic, themessage, mqttpublishqos, mqttpublishretain, function()
	 print("Publish success!!!!!!!!!!!!!") m:close()  m = nil end)  -- function
  end)
end, function(conn,reason) print(reason)
  end) -- function

Here is the traceback (using @nwf dev-ssl branch with extra tls logs:

dofile("test_mqtt2.lua")
enter mqtt_socket_client.
NodeMCU_14a55e
length username: 0
length password: 0
MQTT: Init info: theduke, , 
leave mqtt_socket_client.
enter mqtt_socket_lwt.
mqtt_socket_lwt.
mqtt_socket_lwt: topic: dead, message: iamdead, qos: 0, retain: 0
leave mqtt_socket_lwt.
enter mqtt_socket_connect.
TCP ip is set: 255.255.255.255
TCP port is set: 8883.
Warning, deprecated API! autoreconnect == 1 is deprecatenter socket_dns_found.
TCP ip is set: 198.41.30.241
enter socket_connect.
leave socket_connect, heap = 36400.
leave socket_dns_found.
client handshake start.
TLS<2> (heap=21240): ssl_tls.c:6337 => handshake
TLS<2> (heap=21240): ssl_cli.c:3279 client state: 0
TLS<2> (heap=21240): ssl_tls.c:2416 => flush output
TLS<2> (heap=21240): ssl_tls.c:2428 <= flush output
TLS<2> (heap=21240): ssl_cli.c:3279 client state: 1
TLS<2> (heap=21240): ssl_tls.c:2416 => flush output
TLS<2> (heap=21240): ssl_tls.c:2428 <= flush output
TLS<2> (heap=21240): ssl_cli.c:717 => write client hello
mbedtls_time: 0
TLS<2> (heap=21240): ssl_tls.c:2701 => write record
TLS<2> (heap=21240): ssl_tls.c:2416 => flush output
TLS<2> (heap=21240): ssl_tls.c:2435 message length: 293, out_left: 293
TLS<2> (heap=19624): ssl_tls.c:2441 ssl->f_send() returned 293 (-0xfffffedb)
TLS<2> (heap=19624): ssl_tls.c:2460 <= flush output
TLS<2> (heap=19624): ssl_tls.c:2850 <= write record
TLS<2> (heap=19624): ssl_cli.c:1049 <= write client hello
TLS<2> (heap=19624): ssl_cli.c:3279 client state: 2
TLS<2> (heap=19624): ssl_tls.c:2416 => flush output
TLS<2> (heap=19624): ssl_tls.c:2428 <= flush output
[... 255 lines cut ...]
TLS<2> (heap=19136): ssl_tls.c:2051 <= decrypt buf
TLS<2> (heap=19136): ssl_tls.c:3961 got an alert message, type: [1:0]
TLS<2> (heap=19136): ssl_tls.c:3976 is a close notify message
TLS<1> (heap=19136): ssl_tls.c:3744 mbedtls_ssl_handle_message_type() returned -30848 (-0x7880)
TLS<1> (heap=19136): ssl_tls.c:6578 mbedtls_ssl_read_record() returned -30848 (-0x7880)
TLS<2> (heap=20608): ssl_tls.c:6907 => write close notify
TLS<2> (heap=20608): ssl_tls.c:4032 => send alert message
TLS<2> (heap=20608): ssl_tls.c:2701 => write record
TLS<2> (heap=20608): ssl_tls.c:1258 => encrypt buf
TLS<2> (heap=20608): ssl_tls.c:1560 <= encrypt buf
TLS<2> (heap=20608): ssl_tls.c:2416 => flush output
TLS<2> (heap=20608): ssl_tls.c:2435 message length: 31, out_left: 31
TLS<2> (heap=18992): ssl_tls.c:2441 ssl->f_send() returned 31 (-0xffffffe1)
TLS<2> (heap=18992): ssl_tls.c:2460 <= flush output
TLS<2> (heap=18992): ssl_tls.c:2850 <= write record
TLS<2> (heap=18992): ssl_tls.c:4045 <= send alert message
TLS<2> (heap=18992): ssl_tls.c:6923 <= write close notify
Reason:[-0x7880]
TLS<2> (heap=25152): ssl_tls.c:7065 => free
TLS<2> (heap=37696): ssl_tls.c:7130 <= free
enter mqtt_socket_reconnected.
-5
enter mqtt_socket_disconnected.
leave mqtt_socket_disconnected.
leave mqtt_socket_reconnected.

dtran123, 2017-11-19:
The connection is successful using cloudmqtt. Here is traceback for comparison:

LUA code:

same code as above but replace with cloudmqtt

Logs:

dofile("test_mqtt.lua")
enter mqtt_socket_client.
NodeMCU_14a55e
length username: 5
length password: 5
MQTT: Init info: theduke, xxx, xxx
leave mqtt_socket_client.
enter mqtt_socket_lwt.
mqtt_socket_lwt.
mqtt_socket_lwt: topic: dead, message: iamdead, qos: 0, retain: 0
leave mqtt_socket_lwt.
enter mqtt_socket_connect.
TCP ip is set: 255.255.255.255
TCP port is set: 21938.
Warning, deprecated API! autoreconnect == 1 is deprecated. It will be removed in the next version. See documentation for details.
leave mqtt_socket_connect.
> enter socket_dns_found.
TCP ip is set: 54.234.211.2
enter socket_connect.
leave socket_connect, heap = 34744.
leave socket_dns_found.
client handshake start.
TLS<2> (heap=19592): ssl_tls.c:6337 => handshake
TLS<2> (heap=19592): ssl_cli.c:3279 client state: 0
TLS<2> (heap=19592): ssl_tls.c:2416 => flush output
TLS<2> (heap=19592): ssl_tls.c:2428 <= flush output
TLS<2> (heap=19592): ssl_cli.c:3279 client state: 1
TLS<2> (heap=19592): ssl_tls.c:2416 => flush output
TLS<2> (heap=19592): ssl_tls.c:2428 <= flush output
TLS<2> (heap=19592): ssl_cli.c:717 => write client hello
mbedtls_time: 0
TLS<2> (heap=19592): ssl_tls.c:2701 => write record
TLS<2> (heap=19592): ssl_tls.c:2416 => flush output
TLS<2> (heap=19592): ssl_tls.c:2435 message length: 293, out_left: 293
TLS<2> (heap=17976): ssl_tls.c:2441 ssl->f_send() returned 293 (-0xfffffedb)
TLS<2> (heap=17976): ssl_tls.c:2460 <= flush output
TLS<2> (heap=17976): ssl_tls.c:2850 <= write record
TLS<2> (heap=17976): ssl_cli.c:1049 <= write client hello
TLS<2> (heap=17976): ssl_cli.c:3279 client state: 2
TLS<2> (heap=17976): ssl_tls.c:2416 => flush output
TLS<2> (heap=17976): ssl_tls.c:2428 <= flush output
TLS<2> (heap=17976): ssl_cli.c:1410 => parse server hello
TLS<2> (heap=17976): ssl_tls.c:3728 => read record
TLS<2> (heap=17976): ssl_tls.c:2208 => fetch input
TLS<2> (heap=17976): ssl_tls.c:2366 in_left: 0, nb_want: 5
TLS<2> (heap=17976): ssl_tls.c:2390 in_left: 0, nb_want: 5
TLS<2> (heap=17976): ssl_tls.c:6347 <= handshake
TLS<2> (heap=19592): ssl_tls.c:6337 => handshake
TLS<2> (heap=19592): ssl_cli.c:3279 client state: 2
TLS<2> (heap=19592): ssl_tls.c:2416 => flush output
TLS<2> (heap=19592): ssl_tls.c:2428 <= flush output
TLS<2> (heap=19592): ssl_cli.c:1410 => parse server hello
TLS<2> (heap=19592): ssl_tls.c:3728 => read record
TLS<2> (heap=19592): ssl_tls.c:2208 => fetch input
TLS<2> (heap=19592): ssl_tls.c:2366 in_left: 0, nb_want: 5
TLS<2> (heap=19592): ssl_tls.c:2390 in_left: 0, nb_want: 5
TLS<2> (heap=19592): ssl_tls.c:2391 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
TLS<2> (heap=19592): ssl_tls.c:2403 <= fetch input
TLS<2> (heap=19592): ssl_tls.c:2208 => fetch input
TLS<2> (heap=19592): ssl_tls.c:2366 in_left: 5, nb_want: 66
TLS<2> (heap=19592): ssl_tls.c:2390 in_left: 5, nb_want: 66
TLS<2> (heap=19592): ssl_tls.c:2391 ssl->f_recv(_timeout)() returned 61 (-0xffffffc3)
[... 326 lines cut ...]
TLS<2> (heap=17328): ssl_tls.c:2366 in_left: 0, nb_want: 5
TLS<2> (heap=17328): ssl_tls.c:2390 in_left: 0, nb_want: 5
enter mqtt_socket_sent.
sent1, queue size: 1
Publish success!!!!!!!!!!!!!!!!!!!!!!!
enter mqtt_socket_close.
Send MQTT disconnect infomation, data len: 2, d[0]=224 
TLS<2> (heap=20872): ssl_tls.c:6864 => write
TLS<2> (heap=20872): ssl_tls.c:2701 => write record
TLS<2> (heap=20872): ssl_tls.c:1258 => encrypt buf
TLS<2> (heap=20872): ssl_tls.c:1560 <= encrypt buf
TLS<2> (heap=20872): ssl_tls.c:2416 => flush output
TLS<2> (heap=20872): ssl_tls.c:2435 message length: 31, out_left: 31
TLS<2> (heap=19248): ssl_tls.c:2441 ssl->f_send() returned 31 (-0xffffffe1)
TLS<2> (heap=19248): ssl_tls.c:2460 <= flush output
TLS<2> (heap=19248): ssl_tls.c:2850 <= write record
TLS<2> (heap=19248): ssl_tls.c:6892 <= write
leave mqtt_socket_close.
send_if_poss, queue size: 0
sent2, queue size: 0
leave mqtt_socket_sent.
TLS<2> (heap=23984): ssl_tls.c:7065 => free
TLS<2> (heap=36528): ssl_tls.c:7130 <= free
enter mqtt_socket_disconnected.
leave mqtt_socket_disconnected.

dtran123, 2017-11-19:
I struggled a bit with my personal mosquitto server so that my esp8266 can connect to it via TLS.

It turns out it was my server config that was wrong. If you setup your CA certificates and server certicates properly on your mosquitto server, then the ESP8266 can connect no problem.

Here is my LUA code followed by the logs:

tls.cert.verify(false)
local mqttclientid = "theduke"
local mqtttopic = "sometopic"
local mqttkeepalive = "120"
local mqttuser = "someuser"
local mqttpassword = "somepassword"
local mqttcleansession = "1"
local mqttlwtopic = "dead"
local mqttlwmesssage = "iamdead"
local mqttlwqos = "0"
local mqttlwretain = ""
local mqttserver = "someserver"
local mqttserverport = "8883"
local mqttsecure = "1"
local mqttsubscribeqos = "0"
local mqttpublishqos = "0"
local themessage = "wow super"
local mqttpublishretain = "1"
local m = mqtt.Client(mqttclientid, mqttkeepalive, mqttuser, mqttpassword, mqttcleansession)
m:lwt(mqttlwtopic, mqttlwmesssage, mqttlwqos, mqttlwretain)
m:connect(mqttserver, mqttserverport, mqttsecure, 1, function() m:subscribe(mqtttopic, mqttsubscribeqos, function() m:publish(mqtttopic, themessage, mqttpublishqos, mqttpublishretain, function()
	 print("Publish success!!!!!!!!!!!!!") m:close()  m = nil end)  -- function
  end)
end, function(conn,reason) print(reason)
  end) -- function

logs:

dofile("test_mqtt4_ocw.lua")
enter mqtt_delete.
leave mqtt_delete.
enter mqtt_socket_client.
NodeMCU_14a55e
length username: 5
length password: 5
MQTT: Init info: theduke, ddd, eee
leave mqtt_socket_client.
enter mqtt_socket_lwt.
mqtt_socket_lwt.
mqtt_socket_lwt: topic: dead, message: iamdead, qos: 0, retain: 0
leave mqtt_socket_lwt.
enter mqtt_socket_connect.
TCP ip is set: 122.61.159.199
TCP port is set: 8883.
Warning, deprecated API! autoreconnect == 1 is deprecated. It will be removed in the next version. See documentation for details.
enter socket_connect.
leave socket_connect, heap = 35544.
leave mqtt_socket_connect.
> client handshake start.
TLS<2> (heap=21216): ssl_tls.c:6337 => handshake
TLS<2> (heap=21216): ssl_cli.c:3279 client state: 0
TLS<2> (heap=21216): ssl_tls.c:2416 => flush output
TLS<2> (heap=21216): ssl_tls.c:2428 <= flush output
TLS<2> (heap=21216): ssl_cli.c:3279 client state: 1
TLS<2> (heap=21216): ssl_tls.c:2416 => flush output
TLS<2> (heap=21216): ssl_tls.c:2428 <= flush output
TLS<2> (heap=21216): ssl_cli.c:717 => write client hello
mbedtls_time: 0
TLS<2> (heap=21216): ssl_tls.c:2701 => write record
TLS<2> (heap=21216): ssl_tls.c:2416 => flush output
TLS<2> (heap=21216): ssl_tls.c:2435 message length: 293, out_left: 293
TLS<2> (heap=19600): ssl_tls.c:2441 ssl->f_send() returned 293 (-0xfffffedb)
TLS<2> (heap=19600): ssl_tls.c:2460 <= flush output
TLS<2> (heap=19600): ssl_tls.c:2850 <= write record
[... 333 lines cut ...]
TLS<2> (heap=19000): ssl_tls.c:2366 in_left: 0, nb_want: 5
TLS<2> (heap=19000): ssl_tls.c:2390 in_left: 0, nb_want: 5
enter mqtt_socket_sent.
sent1, queue size: 1
Publish success!!!!!!!!!!!!!
enter mqtt_socket_close.
Send MQTT disconnect infomation, data len: 2, d[0]=224 
TLS<2> (heap=22120): ssl_tls.c:6864 => write
TLS<2> (heap=22120): ssl_tls.c:2701 => write record
TLS<2> (heap=22120): ssl_tls.c:1258 => encrypt buf
TLS<2> (heap=22120): ssl_tls.c:1560 <= encrypt buf
TLS<2> (heap=22120): ssl_tls.c:2416 => flush output
TLS<2> (heap=22120): ssl_tls.c:2435 message length: 31, out_left: 31
TLS<2> (heap=20504): ssl_tls.c:2441 ssl->f_send() returned 31 (-0xffffffe1)
TLS<2> (heap=20504): ssl_tls.c:2460 <= flush output
TLS<2> (heap=20504): ssl_tls.c:2850 <= write record
TLS<2> (heap=20504): ssl_tls.c:6892 <= write
leave mqtt_socket_close.
send_if_poss, queue size: 0
sent2, queue size: 0
leave mqtt_socket_sent.
TLS<2> (heap=25264): ssl_tls.c:7065 => free
TLS<2> (heap=37808): ssl_tls.c:7130 <= free
enter mqtt_socket_disconnected.
leave mqtt_socket_disconnected.

nwf, 2017-11-19:
@dtran123 The first of those messages looks like a normal TLS exchange.  The server is closing the connection -- and sending a close-notify alert -- immediately after you send your credentials.  Are you sure you are using the correct credentials?

In any case, the failure mode is different in your logs ("client handshake ok!") than in the initial bug report ("client handshake failed!") and so I think you may be barking up the wrong tree.

dtran123, 2017-11-19:
Well the fact that TLS connections worked for cloudmqtt and for my personal mosquitto server configured with self signed certifcations, I am confident that there was probably something wrongly configured for the iot.eclipse.org scenario. But I am finally happy to see that we can get basic secured mqtt connections !

closed 2018-02-24 by marcelstoer