nodemcu/nodemcu-firmware #2073: uart feature

ShAzmoodeh, 2017-08-14:
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature
limited input charecter in uart.on

### Justification
many programs need to receive above charecter than 255

8<------------------------ END FEATURE REQUEST --------------------------------

nwf, 2017-08-15:
For memory-constrained devices like the esp8266, you are likely much better off doing incremental processing of your communications stream.  If you absolutely must accumulate strings larger than 255 characters, you can do so manually in Lua.

ShAzmoodeh, 2017-08-15:
yeh . i do it and no pb . but if it featuer contanied into the firmware it might be better 
now i stream a string in lua ,thats larger than 255 charecter and make a chank of number of strings then workon strings . 
just may sometimes esp go to reset mode and i think its occur because we using micro lua language with limited resource or ported lua for this module perhaps i write ugly code in lua(my first lang is c)

TerryE, 2017-10-20:
Another good reason for sticking to the 255  char limit is that most USB connected virtual uarts have a 256 byte internal buffer.  It's a _lot_ safer doing some form of S/W handshake / flow control when doing hi-volume xfer over the UART.

closed 2018-02-24 by marcelstoer