nodemcu/nodemcu-firmware #2074: ESP32: netconn error

LDmitryL, 2017-08-14:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Expected transfer with no errors
### Actual behavior
Sometimes when transferring files an error occurs:
'PANIC: unprotected error in call to Lua API (attempt to call a nil value)'
Without the line number program code. But this always occurs after the event "receive".
Sometimes these actions result in an error:
`sdk/esp32-esp-idf/components/freertos/heap_regions.c:368 (vPortFreeTagged)- assert failed!
abort() was called at PC 0x4008597f on core 0`
These events are rare. You can spend hours doing the same queries and that does not happen. But sometimes the same query leads to one of the described errors.

### Test code
```Lua
local reqTypes = {GET = true,POST = false};
local extmap = {txt="text/plain",html="text/html",htm="text/html",lua="text/plain",js="application/javascript",css="text/css"};
reqdata = {}; openF=0;

local function sendFC()
    local line,buf, status, error
print("send packet for",n,SS)
    if (openF==1) then
            line=file.readline();
            if line then 
                buf=("%X\r\n"):format(#line)..line.."\r\n";
                status, error =pcall(function() SS:send(buf, function (c) sendFC(); end) end); 
                if status == false then print("Send error: ",error,", file",n,"from",SS); SS:close(); file.close(); end
            else
print("close file",n,"from",SS);
                status, error =pcall(function () SS:send("0\r\n\r\n"); end);  
                if (status == false) then  print("Send error END of chunked",error); end;
                file.close(); openF=2; reqdata = {}; 
                status, error =pcall(function () SS:close(); end);  if (status == false) then  print("close error",error); end;
print("close connection",SS); openF=3;  collectgarbage("collect");
            end 
    end
end 

sendFileContents = function(conn, Type, Name)
print("SEND", Name, "FROM", conn)
    n=Name;    SS=conn; -- for debugging use via global variables
    file.open(Name, "r");
    openF=1;   --keep track of the open file
    sendFC();
end 

responseHeader = function(code, type)
local cache="\r\nCache-Control: max-age=31536000\r\n";
return "HTTP/1.1 "..code.." \r\nServer: luaweb\r\nContent-Type: "..type..cache.."Transfer-Encoding: chunked\r\nConnection: close\r\n\r\n" 
end

receiver=function(conn,request)
print("-------- BEGIN --------");
print("-- request ->\n[[",request,"]]")
            local ftype, fname, status, error, line, rest, s, e, method,req,major,minor;
            _, _, method, req, major, minor = string.find(request, "([A-Z]+) (.+) HTTP/(%d).(%d)");
            if reqTypes[method] then
                fname = ""; --local
                if req:find("%?") then
-- GET
                    _, _, fname, rest = req:find("(.*)%?(.*)");
                    rest = rest .. "&";
                    for crtpair in rest:gmatch("[^&]+") do
                        local _, _, k, v = crtpair:find("(.*)=(.*)");
                    -- replace all "%xx" characters with their actual value
                        v = v:gsub("(%%%x%x)", function(s) return string.char(tonumber(s:sub(2, -1), 16)) end);
                        reqdata[k] = v; 
                    end
                else
-- POST (to do later)
                    fname = req;
                end
                fname = ( fname == "/" ) and "index.html" or fname:sub(2, -1);
                s, e = fname:find("%.[%a%d]+$");
                --ftype=""; -- local
                if s~=nil then ftype = fname:sub(s+1, e):lower(); else ftype='html'; fname=fname:lower()..'.html'; end
                ftype = (#ftype > 0 and ftype) or "txt"; 
                if file.exists(fname) then
                    status, error = pcall(function() conn:send(responseHeader("200 OK",extmap[ftype or "txt"]) ); end); 
                    if (status == false) then print("Send header error, close! " .. error); else sendFileContents(conn,ftype,fname); end
                else
                    line="<html><head><title>Error</title></head><body><h1>Page not found</h1</body></html>";
                    line=("%X\r\n"):format(#line)..line.."\r\n0\r\n\r\n"
                    line=responseHeader("404 Not Found","text/htm")..line;
                    status, error = pcall(function() conn:send(line); end );
                    status, error = pcall(function() conn:close(); end);  if (status == false) then  print("close error",error); end;
                end
                fname, ftype ,s, e = nil,nil,nil,nil;
            else
                line="<html><head><title>Error</title></head><body><h1>400 Invaild Request</h1</body></html>";
                status, error = pcall(function() conn:send(responseHeader("400 Bad Request","text/htm")..("%X\r\n"):format(#line)..line.."\r\n0\r\n\r\n"); end);
                status, error = pcall(function() conn:close(); end);  if (status == false) then  print("close error",error); end;
            end
print("------ END --------");
end

collectgarbage("setmemlimit", 64);
srv=net.createServer(net.TCP,30);
srv:listen(80, function(S)
        S:on("receive", function(con,reg) receiver(con,reg); end )
        S:on("disconnection",function()  print("[[ disconnect ]] "); openF=0; collectgarbage("collect");  end)
        S:on("sent",function() print("[==[ sent ]==]"); end)
          end);
```

### NodeMCU version
the latest version of netconnect

### Hardware
ESP Wroom32
8<------------------------ END BUG REPORT -------------------------------------

LDmitryL, 2017-08-14:
@devsaurus 
Most often the errors occur when accessing via the AP. But using STA also happen. When you request every 30 seconds during the day bugs can be 3-4 times. But can crumble in a row. Often the error like this:
```
[[
GET /style.min.css HTTP/1.1
Host: 192.168.4.1
Connection: keep-alive
User-Agent: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/59.0.3071.125 Safari/537.36
Accept: text/css,*/*;q=0.1
Referer: http://192.168.4.1/
Accept-Encoding: gzip, deflate
]]

SEND	style.min.css	FROM	userdata: 0x3ffcd380
send packet, type=	false	for	style.min.css	userdata: 0x3ffcd380
------ END --------
```

`PANIC: unprotected error in call to Lua API (attempt to call a nil value)`

If you look at the code the error occurs at the time of the event  ` ------ END --------`

Maybe it's because of the transfer of binary data (not TXT)? I give the compressed GZIP data.
Sorry more I can not answer about this error. Error occurs rarely. And I can't localize the source of error.

LDmitryL, 2017-08-14:
@devsaurus The most interesting thing that has arisen when modifying the code sendFileContents as recommended (using a callback). I sent the answer in a loop (while not EOF file). In this case, the transmission time of all content was 4 seconds. And with the use of technology callback 20 seconds.

devsaurus, 2017-08-14:
@LDmitryL you open up a race condition in your code with the consecutive `:send()` and `:close()` commands. Either the sequence works as you intend (send data is transmitted and socket is closed afterwards) or the socket is closed while sending is in progress which might break the lwip stack at this point. I've added some result code checks to `socket:close()` in 412db2e3391e4769fbe72d508b43522f7b70b2d1, so you should be notified when you lose the race.

In any case, I recommend to move the `:close()` into the sent callback like in this example:
```lua
SS:send("0\r\n\r\n", function (c)
  print("sending done, closing connection")
  c:close()
end)
```

An unrelated note: You use the global variable `SS` to store the current socket. This will break for parallel incoming connections. I'd propose to avoid this global and transport the socket as parameter to `sendFC()` instead.

marcelstoer, 2017-08-14:
@LDmitryL our issue template asks for a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) - yours ain't minimal at all. Please help the folks here to spend their time more effectively to reduce the amount of code one has to wade through.

closed 2017-08-14 by marcelstoer