nodemcu/nodemcu-firmware #2076: Invalid hostname when use wifi.sta.sethostname

hanfengcan, 2017-08-15:
hi

I use the Latest release, I can't sethostname use the code 
`
if (wifi.sta.sethostname("NodeMCU") == true) then
    print("hostname was successfully changed")
else
    print("hostname was not changed")
end
`

It said `bad argument #1 to 'sethostname' (Invalid hostname)`

what went wrong

marcelstoer, 2017-08-15:
That example in our documentation is bogus, sorry. The documentation says that the function returned `nil` rather than boolean:

https://nodemcu.readthedocs.io/en/latest/en/modules/wifi/#wifistasethostname

```
wifi.sta.sethostname("NodeMCU-you")
print(wifi.sta.gethostname())
```

prints "NodeMCU-you". If you have invalid chars in the hostname (check for non printable chars) it does indeed report "Invalid hostname".

hanfengcan, 2017-08-16:
What a pity!:joy:

use `wifi.sta.sethostname("NodeMCU-you")`, is also said `bad argument #1 to 'sethostname' (Invalid hostname)` :sweat_smile:

It is some wrong in the firmware? My code can run before I updated firmware:dizzy_face:

marcelstoer, 2017-08-16:
> I use the Latest release

Is not very precise. What do you see when NodeMCU boots? Where did you get the firmware from?

dnc40085, 2017-08-17:
My mistake, I must have forgotten to add the return.
Here is a PR that should fix this.

dnc40085, 2017-08-17:
@hanfengcan The pull request would first have to be merged. Afterwards, you would be able to build firmware from the dev branch, which would then contain the fix.

If you are in a hurry and need a fixed version of the firmware, I can build one for you if you provide the list of modules you need and whether you need the float or integer version.

hanfengcan, 2017-08-17:
@dnc40085  OK, thanks

satech-domo, 2017-09-02:
Hi,

I got a variant of this problem:
I just made a new firmware before testing and clear all emory
	NodeMCU custom build by frightanic.com
	branch: master
	commit: 2e67ff5a639a13260fd4fb3c3b627ccdc2845616
	SSL: false
	modules: adc,bit,file,gpio,i2c,net,node,tmr,uart,wifi
 	build 	built on: 2017-09-02 20:46
 	powered by Lua 5.1.4 on SDK 2.1.0(116b762)

I did on ESP8266 at 192.168.0.171:
- get HostName before: NODE-33C58A
- set HostName : ESP-01
- get HostName after: ESP-01
the code is:
	-- Set Network Host Name
	local hy = wifi.sta.gethostname()
	print('get HostName before: ' .. hy)		
	print('set HostName : ' .. "ESP-01")
	wifi.sta.sethostname("ESP-01")
	local hz = wifi.sta.gethostname()
	print('get HostName after: ' .. hz)				

so things looks OK,
but when I do an Advanced IP Scanner I get HostName = IP adr. not ESP-01
Name			        IP
192.168.0.171		        192.168.0.171
NODE-33C58A.local	192.168.0.174
No response when ping ESP-01
On another ESP at ip 174 the name is the orignal set by Expressif.

Salutations

marcelstoer, 2017-09-02:
> No response when ping ESP-01

That's primarily a problem of the component which is responsible for name resolution in your network.

satech-domo, 2017-09-04:
Thank for quick response

 

I will setup with another router, to see if name resolution will be conform.

Salutations

PR #2079 (2017-09-15): Add missing return value for wifi.sta.sethostname
Fixes #2076.

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested my contribution.
- [X] The code changes are reflected in the documentation at `docs/en/*`.

This PR adds the missing Boolean return value to `wifi.sta.sethostname()`. Also, some refactoring was performed to allow SDK function return value to be forwarded to the de [... cut]
closed 2017-09-15 by marcelstoer