nodemcu/nodemcu-firmware #2077: ESP-01 doesn't stop restarting

MohammedBENHADINE, 2017-08-15:
Hi Makers,
I did upload a lua firmware to my ESP-01 and i tried to upload some lua scripts to test the diffrent modules & features.
The problem is that many times it starts reseting without any reason (even when i delete all the scripts from the Flash memory).
Please let me know if i'am mistaking something about the firmware version
Thanks in advance
8<------------------------ BUG REPORT -----------------------------------------

### Actual behavior
NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4
lua: cannot open init.lua
> c_ÇÏRSöfJSúfJÃjîè

NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4
lua: cannot open init.lua
> c_ÇRSöfJSúfJÃjîè

NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4
lua: cannot open init.lua
> c_ÇÏRSö&JSúfJÃîè

### Test code

### NodeMCU version
NodeMCU 0.9.6
### Hardware
ESP-01

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ END FEATURE REQUEST --------------------------------

MohammedBENHADINE, 2017-08-15:
`m_dis={}
function dispatch(m,t,pl)
	if pl~=nil and m_dis[t] then
		m_dis[t](m,pl)
	end
end
function topic1func(m,pl)
	print("get1: "..pl)
end
function topic2func(m,pl)
	print("get2: "..pl)
end
m_dis["/topic1"]=topic1func
m_dis["/topic2"]=topic2func
-- Lua: mqtt.Client(clientid, keepalive, user, pass)
m=mqtt.Client("nodemcu1",60,"bridge","lora")
m:on("connect",function(m) 
	print("connection "..node.heap()) 
	m:subscribe("/topic1",0,function(m) print("sub done") end)
	m:subscribe("/topic2",0,function(m) print("sub done") end)
	m:publish("/topic1","hello",0,0) m:publish("/topic2","world",0,0)
	end )
m:on("offline", function(conn)
    print("disconnect to broker...")
    print(node.heap())
end)
m:on("message",dispatch )
-- Lua: mqtt:connect( host, port, secure, auto_reconnect, function(client) )
m:connect("192.168.43.205",1883,0,1)
tmr.alarm(0,10000,1,function() local pl = "time: "..tmr.time() 
	m:publish("/topic1",pl,0,0)
	end)`
this 's my testing.lua script

TerryE, 2017-08-15:
See #1010.  Also use a current S/W version and read our the document.

closed 2017-08-15 by TerryE