nodemcu/nodemcu-firmware #2080: Cron not working for latest release

abhignair, 2017-08-17:
**Please note that the cron module works using build from branch 2.1.0-master_20170521 but doesn't work using the latest release (master branch) from the cloud build service.**

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Output using nodemcu firmware from branch 2.1.0-master_20170521.

NodeMCU 2.1.0 build unspecified powered by Lua 5.1.4 on SDK 2.1.0(116b762)
> Time Sync  2017/08/17 08:32:21
Every Minute: 2017/08/17 08:32:21
Every Minute: 2017/08/17 08:33:00
Every Minute: 2017/08/17 08:34:00

...and so on with output of same format as above with different time every minute on new line.

### Actual behavior
Output using nodemcu firmware from the cloud build service.

NodeMCU custom build by frightanic.com
	branch: master
	commit: b62fae918b3627209fe434950cffb2506cee9a31
	SSL: false
	modules: cron,file,gpio,net,node,rtctime,sntp,tmr,uart,wifi
 build 	built on: 2017-08-17 06:35
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
> Time Sync  2017/08/17 07:16:31

...and no further output.
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
function sync_success()
    tm = rtctime.epoch2cal(rtctime.get())
    print(string.format("Time Sync  %04d/%02d/%02d %02d:%02d:%02d", tm["year"], tm["mon"], tm["day"], tm["hour"], tm["min"], tm["sec"]))
    cron.schedule("* * * * *", function(e) 
		      tm = rtctime.epoch2cal(rtctime.get())
		      print(string.format("Every Minute: %04d/%02d/%02d %02d:%02d:%02d", tm["year"], tm["mon"], tm["day"], tm["hour"], tm["min"], tm["sec"]))
    end)
end

wifi.eventmon.register(wifi.eventmon.STA_GOT_IP, sntp.sync(nil, sync_success, nil, nil))
    
wifi.setmode(wifi.STATION)
    
station_cfg={}
station_cfg.ssid="xxxx"
station_cfg.pwd = "xxxx"
wifi.sta.config(station_cfg)
wifi.sta.connect()
```
### NodeMCU version
The build for which the above code doesn't work is from the master branch from the cloud build website (https://nodemcu-build.com/index.php) so the branch must be 2.1.0-master_20170811.

**Please note that my code works for the previous release of branch 2.1.0-master_20170521.**
I used Docker to build the previous release.

### Hardware
I am using the D1 mini with no special hardware setup.

8<------------------------ END BUG REPORT -------------------------------------

nwf, 2017-08-17:
As you are able to build your own firmwares, could you try 'git bisect'-ing this for us, to narrow it down to a particular commit?

ritesht93, 2017-08-19:
I am facing the similar issue. Last commit where cron worked properly for me was c8ac5cfb912ff206b03dd7c60ffbb2dafb83fe5e.

pazulin, 2017-08-22:
I meet same problem. cron is not working .  now I use older version

nwf, 2017-08-22:
@ritesht93 Do you mean that e7f063950 doesn't work for you?

dnc40085, 2017-08-23:
@nwf git bisect output:
```
d93465cd865dc21efac277e920cf7169d88273d7 is the first bad commit
commit d93465cd865dc21efac277e920cf7169d88273d7
Author: Philip Gladstone <pjsg@users.noreply.github.com>
Date:   Tue Jul 18 16:51:20 2017 -0400

    Add tracking and control of the rate error in the clock crystal. (#1697)
    
    * Improve RTC timekeeping -- includes clock rate tracking
    * Improved division by 1M
    * Fix crash in sntp
    * Disable RTC debug
    * Get the offset correct
    * Add comments on where the mysterious numbers came from
    * Fix a crash with auto repeat mode and errors on repeat

:040000 040000 cbae4a4c4513941eb16daa0af53e11232aa9277b 92155b21adc0d1cb380b825638cd59042c39d675 M	app
:040000 040000 3114c2c327532399ebe0c4affb189418a24254ff 0d9901491740cca399d46121c618f2a3f480eda8 M	docs
```

marcelstoer, 2017-08-24:
Thanks for reporting @abhignair.

commit f9e0aab2 (2017-08-24): cron: don't include rtctime_internal.h (#2084)

That file is supposed to only be included once because it does things
like declare static globals.  As it stands, cron doesn't believe time is
ticking.

Fixes #2080
closed 2017-08-24 by marcelstoer