nodemcu/nodemcu-firmware #2081: NodeMCU Deauther

sarimnaveed2, 2017-08-17:
Can anyone tell me what "probe request" attack does on NodeMCU Deauther ?
 

_Sent from my Motorola XT1060 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

nwf, 2017-08-17:
You seem confused.  I think you are referring to https://github.com/spacehuhn/esp8266_deauther , but while that uses the ESP8266, it does not use the NodeMCU software except, apparently, for the firmware flash tool on Windows.

sarimnaveed2, 2017-08-18:
Yes I meant esp8266

_Sent from my Motorola XT1060 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

closed 2017-08-18 by marcelstoer

devyte, 2017-08-18:
This is the repository for the Lua based firmware. It has very little to do
with the Arduino based firmware, and the libraries are also not compatible.
Please close this.

Also, these are issue trackers, meant for tracking issues in the firmware,
and not for asking questions. I suggest directing your question at a
discussion forum like esp8266.com or stackoverflow.