nodemcu/nodemcu-firmware #2082: wifi.sta.config("SSSID","PASSWORD") not working anymore

MacSass, 2017-08-20:
Hi, 
until Version 2.1.0-master_20170521 I was able to set the WIFI AP configuration with wifi.sta.config("SSSID","PASSWORD"), which for example is used by the ESP8266 LuaLoader 0.91. (conveniently hardcoded to a button)

Since the new version 2.1.0-master_20170811 this command returns an error with:
_stdin:1: bad argument #1 to 'config' (config table not found!)
stack traceback:
	[C]: in function 'config'
	stdin:1: in main chunk_

Obviously I´m still able to set the wifi AP with:
_station_cfg={}
station_cfg.ssid="SSSID"
station_cfg.pwd="Password"
station_cfg.save=true
wifi.sta.config(station_cfg)_

But I found the old version easier to type and it is hardcoded to the LuaLoader button.

Has this change been made intentionally?

Thanks - MacSass

### Test code

```Lua
wifi.sta.config("SSSID","PASSWORD")
```
### NodeMCU version
build 	built on: 2017-08-18 21:41
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
2.1.0-master_20170811

### Hardware
esp8266

marcelstoer, 2017-08-20:
> Has this change been made intentionally?

Yes, back in April we added a deprecation message, see #1939. I created https://github.com/GeoNomad/LuaLoader/issues/5

closed 2017-08-20 by marcelstoer

MacSass, 2017-08-21:
Thanks Marcel for your fast feedback - I missed that deprecation message before.
I´ll contact the author of LuaLoader in hope he does an update (he hasn´t for a very long time).
Thanks again -MacSass

TerryE, 2017-08-21:
BTW, how about:
```Lua
wifi.sta.config {ssid="SSSID", pwd="Password" }
```
Not too painful, neh?

MacSass, 2017-08-22:
Hi Terry, 
thanks for the note - this is in fact a nice way that I was not aware of (my lack of knowledge on lua).

Still the LuaLoader would need an update - but binding your command to the custom button for sure is a nice workaround for now.
Unfortunately the orignal author of LuaLoader does not have time to update and I don´t have any Delphi Know-How ...

Thanks - MacSass

marcelstoer, 2017-08-22:
I identified the code that likely needs to change (see my first comment). Maybe you can help the author with a PR.

MacSass, 2017-08-22:
Yeah,
I´ll try that - fairly new to GitHub as well - so might be a good learning for me anyhow.

Thanks - MacSass

syadav8126, 2017-09-04:
station_cfg={}
station_cfg.ssid="SSSID"
station_cfg.pwd="Password"
station_cfg.save=true
wifi.sta.config(station_cfg)
still cant connect to a AP, so please suggest me a solution.
terminal shows
bad argument #1 to config(config table not found)

kirkryan, 2017-09-13:
When I run the following code:

`print(wifi.sta.getip())
wifi.setmode(wifi.STATION)
wifi.sta.config {ssid="MyWiFi", pwd="123456789"}
print(wifi.sta.getip())`

I get the following:

`> dofile('')
cannot open 
stack traceback:
	[C]: in function 'dofile'
	stdin:1: in main chunk`

I'm a total noob to this and all the tutorials use the old command so any assistance would be greatly appreciated!

marcelstoer, 2017-09-13:
> I'm a total noob to this and all the tutorials use the old command so any assistance would be greatly appreciated!

That's what https://nodemcu.readthedocs.io/en/latest/en/support/ is for.