nodemcu/nodemcu-firmware #2083: Can't run more than once. Can't print anything.

zszen, 2017-08-22:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
1.print words.
2.Can repeat save and run again.

### Actual behavior
1.Don't display print words. 
2.Save again don't run again.  "Waiting answer from ESP - Timeout reached. Command aborted."

### Test code
connection info  :
NodeMCU custom build by frightanic.com
	branch: master
	commit: b62fae918b3627209fe434950cffb2506cee9a31
	SSL: false
	modules: adc,ads1115,am2320,bit,cron,crypto,encoder,file,gpio,i2c,net,node,ow,pcm,pwm,rtctime,sjson,spi,struct,switec,tmr,u8g,uart
 build 	built on: 2017-08-22 00:26
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)

zszen, 2017-08-22:
I use it control 12-pin digital display. It maybe convert tx/rx to gpio control. So each test I must reset the chipboard. And luck I don't run it as init.lua. It will never be written in. Other wise format it.
[image]

marcelstoer, 2017-08-22:
See #1010

closed 2017-08-22 by marcelstoer