nodemcu/nodemcu-firmware #2087: Building ESP32

crishead, 2017-08-24:
I made a build from repository on Linux (64 bit). I set the modules (make menuconfig).
The build was ok, and the flash too, but the result not work fine:
gpio.mode(4,1)
stdin:1: attempt to call field 'mode' (a nil value)
=print(gpio)
romtable: 0x3f403970 //it's OK

=print(gpio.OUTPUT)
nil

The enviroment:
Linux x64
GNU Make 4.1
----------------------------
I made a Debian x86 virtual machine, and I made this steps again, but the result was same.

the boot text from Esplorer:

Communication with MCU..Got answer! Communication with MCU established.
AutoDetect firmware...

Can't autodetect firmware, because proper answer not received (may be unknown firmware). 
Please, reset module or continue.
ts Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:4428
ho 0 tail 12 room 4
load:0x40078000,len:11072
load:0x40080000,len:252
entry 0x40080034
[0;32mI (48) boot: ESP-IDF v3.0-dev-20-g9b955f4c 2nd stage bootloader[0m
[0;32mI (48) boot: compile time 08:22:42[0m
[0;32mI (49) boot: Enabling RNG early entropy source...[0m
[0;32mI (65) boot: SPI Speed      : 40MHz[0m
[0;32mI (78) boot: SPI Mode       : DIO[0m
[0;32mI (90) boot: SPI Flash Size : 4MB[0m
[0;32mI (103) boot: Partition Table:[0m
[0;32mI (114) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (137) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (160) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (183) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (207) boot:  3 spiffs           unknown          c2 00 00110000 002f0000[0m
[0;32mI (230) boot: End of partition table[0m
[0;32mI (243) boot: Disabling RNG early entropy source...[0m
[0;32mI (260) boot: Loading app partition at offset 00010000[0m
[0;32mI (1290) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) [0m
[0;32mI (1291) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x16f2c ( 93996) map[0m
[0;32mI (1308) boot: segment 2: paddr=0x00036f3c vaddr=0x3ffb0000 size=0x01f34 (  7988) load[0m
[0;32mI (1337) boot: segment 3: paddr=0x00038e78 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (1361) boot: segment 4: paddr=0x00039280 vaddr=0x40080400 size=0x13de4 ( 81380) load[0m
[0;32mI (1426) boot: segment 5: paddr=0x0004d06c vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (1427) boot: segment 6: paddr=0x0004d0d8 vaddr=0x00000000 size=0x02f30 ( 12080) [0m
[0;32mI (1446) boot: segment 7: paddr=0x00050010 vaddr=0x400d0018 size=0x70e6c (462444) map[0m
[0;32mI (1472) cpu_start: Pro cpu up.[0m
[0;32mI (1483) cpu_start: Starting app cpu, entry point is 0x40081094[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1516) heap_alloc_caps: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1539) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM[0m
[0;32mI (1559) heap_alloc_caps: At 3FFB7720 len 000288E0 (162 KiB): DRAM[0m
[0;32mI (1580) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1602) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1623) heap_alloc_caps: At 400941E4 len 0000BE1C (47 KiB): IRAM[0m
[0;32mI (1644) cpu_start: Pro cpu start user code[0m
[0;32mI (202) cpu_start: Starting scheduler on APP CPU.[0m

Mounting flash filesystem...
I (412) wifi: wifi firmware version: 72ddf26
I (412) wifi: config NVS flash: enabled
I (412) wifi: config nano formating: disabled
[0;32mI (412) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (422) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (452) wifi: Init dynamic tx buffer num: 32
I (452) wifi: Init dynamic rx buffer num: 32
I (452) wifi: wifi driver task: 3ffbf920, prio:23, stack:4096
I (452) wifi: Init static rx buffer num: 10
I (462) wifi: Init dynamic rx buffer num: 32
I (462) wifi: Init rx ampdu len mblock:7
I (462) wifi: Init lldesc rx ampdu entry mblock:4
I (472) wifi: wifi power manager task: 0x3ffc4ce8 prio: 21 stack: 2560
NodeMCU ESP32 build unspecified powered by Lua 5.1.4
lua: cannot open init.lua
> 
=node.heap()
=node.heap()
242600
> 
=
=
> =node.chipid()
0xb3240ac412fd
> =node.info()
stdin:1: attempt to call field 'info' (a nil value)
> =node.flashid()
stdin:1: attempt to call field 'flashid' (a nil value)
> 
=print(GPIO)
=print(GPIO)
nil
> 
=print(gpio)
=print(gpio)
romtable: 0x3f403970
> 
=print(gpio.OUTPUT)
=print(gpio.OUTPUT)
nil
> 
gpio.mode(4,1)
gpio.mode(4,1)
stdin:1: attempt to call field 'mode' (a nil value)
> =print(gpio)
=print(gpio)
romtable: 0x3f403970
> =print(gpio.OUTPUT)
=print(gpio.OUTPUT)
nil
----------------------------------------------
Any ideas?

larsstenberg, 2017-08-25:
esp32 GPIO module is incompatible with ESP8266 (master) setup.

Check documentation for the esp32 branch: https://nodemcu.readthedocs.io/en/dev-esp32/en/modules/gpio/

crishead, 2017-08-29:
Oh, yes, sorry, I was careless.
Thank you!

closed 2017-08-29 by crishead