nodemcu/nodemcu-firmware #2089: Interacting issues with ESP-01

MohammedBENHADINE, 2017-08-25:
Hi guys,
I'm getting problems with the intercation with my esp8266,
After flashing the hardware with a newest firmware(from the could build) ,,i'm incapable sending some lua snippets (i cannot get any answer from the device).
I did tried with many baudrates other than 9600 & 115200.
I verified the wiring,and the current is quite enough for the module(external source).
Could you please help !

### Actual behavior
NodeMCU custom build by frightanic.com
	branch: dev
	commit: f9e0aab200e974116c569e6241f4270e8afbe637
	SSL: false
	modules: file,gpio,net,node,tmr,uart,websocket,wifi
 build 	built on: 2017-08-25 15:56
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
lua: cannot open init.lua
> 
=node.heap()
=nÿ
=node.heap()
=node.heap()
=þ
=þ
=node.heap()
=þ
=wifi.ap.getip()
=÷
=wifi.ap.getip()
=ÿ
=wifi.sta.getmac()
=÷
=wifi.sta.getmac()
=ÿ
### NodeMCU version
The latest (SDK 2.1.0)

### Hardware
ESP-01
8<------------------------ END BUG REPORT -------------------------------------

marcelstoer, 2017-08-25:
See #1010

closed 2017-08-25 by marcelstoer

MohammedBENHADINE, 2017-08-25:
Thanks a lot for this awesome answer

TerryE, 2017-08-26:
@MohammedBENHADINE Mohammed, our point is that there is an appropriate forum for different classes of Qs.  Here is specifically specific bugs and enhancements in the firmware.  StackOverflow, esp8266.com, etc. is where you seek this sort of help.