nodemcu/nodemcu-firmware #2094: Cron module doesn't work

ignatievvictor, 2017-09-01:
### Expected behavior
I want to use a cron schedule. Trying code from docs. 
It should print test string every minute

### Actual behavior
Nothing happens. I wait much more than 1 min.
Have reload code several times. Have tried another type of code.

### Test code

```Lua
function my_schedule() 
  tm = rtctime.epoch2cal(rtctime.get())
  print("It's Cron.")
  print(string.format("%04d/%02d/%02d %02d:%02d:%02d", tm["year"], tm["mon"], tm["day"], tm["hour"], tm["min"], tm["sec"])) 
end
 
-- Cron. Every minute 
cron.schedule("* * * * *", my_schedule)
my_schedule() 
```
### NodeMCU version
This was built against the dev branch and includes the following modules: bit, cron, dht, file, gpio, http, i2c, net, node, rfswitch, rtctime, spi, tmr, uart, websocket, wifi.

i've tried master and dev both. 
[image]

### Hardware

NodeMCU V2

djphoenix, 2017-09-01:
Hello onboard.

Does your app have RTC time source (e.g. configured SNTP via sntp.sync call, or another part that calls rtctime.set)?

ignatievvictor, 2017-09-01:
I have a RTC module (shield with a battery)
Oh my! should i need call a rtctime.set before use a cron??

ignatievvictor, 2017-09-01:
djphoenix, thank you, my friend.
I just check it and it works like a charm!
I must do rtctime.set(timestamp, 0) before use a cron.
I think this trick should be reflected in official docs in cron module.

closed 2017-09-01 by ignatievvictor

djphoenix, 2017-09-01:
> should be reflected in official docs in cron module

Of course! And we [did it](http://nodemcu.readthedocs.io/en/dev/en/modules/cron/) :)
> This module needs RTC time to operate correctly. Do not forget to include the rtctime module and initialize it properly.