nodemcu/nodemcu-firmware #2095: wifi.sta.config() problem

ab988, 2017-09-04:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
accept ssid and password
### Actual behavior
show this message:
lua: init.lua:18: bad argument #1 to 'config' (config table not found!)
stack traceback:
	[C]: in function 'config'
	init.lua:18: in main chunk
	[C]: ?

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
```
### NodeMCU version
modules: file,gpio,net,node,tmr,uart,wifi
 build 	built on: 2017-09-03 20:53
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)

i was test on 0.9.5 version and dont have any problem.

thank

marcelstoer, 2017-09-04:
The API changed recently: https://nodemcu.readthedocs.io/en/latest/en/modules/wifi/#wifistaconfig

closed 2017-09-04 by marcelstoer

ab988, 2017-09-05:
thanks
i have change code as below not show any error 
and connect to medem but when i writing ip do not show web page

wifi.setmode(wifi.STATION)
station_cfg={}
station_cfg.ssid="my ssid"
station_cfg.pwd="password"
wifi.sta.config(station_cfg)
wifi.sta.connect()
print(wifi.sta.getip())
led1 = 3
led2 = 4
gpio.mode(led1, gpio.OUTPUT)
gpio.mode(led2, gpio.OUTPUT)
srv=net.createServer(net.TCP)
srv:listen(80,function(conn)
    conn:on("receive", function(client,request)
        local buf = "";
        local _, _, method, path, vars = string.find(request, "([A-Z]+) (.+)?(.+) HTTP");
        if(method == nil)then
            _, _, method, path = string.find(request, "([A-Z]+) (.+) HTTP");
        end
        local _GET = {}
        if (vars ~= nil)then
            for k, v in string.gmatch(vars, "(%w+)=(%w+)&*") do
                _GET[k] = v
            end
        end
        buf = buf.."<h1> ESP8266 Web Server</h1>";
        buf = buf.."<p>GPIO0 <a href=\"?pin=ON1\"><button>ON</button></a>&nbsp;<a href=\"?pin=OFF1\"><button>OFF</button></a></p>";
        buf = buf.."<p>GPIO2 <a href=\"?pin=ON2\"><button>ON</button></a>&nbsp;<a href=\"?pin=OFF2\"><button>OFF</button></a></p>";
        local _on,_off = "",""
        if(_GET.pin == "ON1")then
              gpio.write(led1, gpio.HIGH);
        elseif(_GET.pin == "OFF1")then
              gpio.write(led1, gpio.LOW);
        elseif(_GET.pin == "ON2")then
              gpio.write(led2, gpio.HIGH);
        elseif(_GET.pin == "OFF2")then
              gpio.write(led2, gpio.LOW);
        end
        client:send(buf);
        client:close();
        collectgarbage();
    end)
end)

thanks

marcelstoer, 2017-09-05:
This is not the right place for user support, see #1010.