nodemcu/nodemcu-firmware #2096: ESP:32 Bug in net.server:close()

LDmitryL, 2017-09-07:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Closed connection at the end of transmission
### Actual behavior
Error in the performance of functions `c:close()` and shutdown of the system
The client part receives the test message and closes the connection without error.

Client
>Connected to 192.168.2.66
Sent GET / HTTP/1.1
Received
TEST LINE
Connection closed

The LUA console
>TEST LINE
	con=	userdata: 0x3ffdcfd8
sending done, closing connection	userdata: 0x3ffdcfd8
PANIC: unprotected error in call to Lua API (stdin:7: illegal value)

### Test code
```Lua
srv=net.createServer(net.TCP,30);
srv:listen(66666, function(S)
                S:on("receive", function(con,reg) 
                    line='\nTEST LINE\n';
                    print(line,"con=",con)
                    con:send(line,function (c) 
                                            print("sending done, closing connection",c); 
                                            c:close(); 
                                  end); 
                end); 
end);
```
### NodeMCU version
NodeMCU ESP32 build unspecified powered by Lua 5.1.4
ESP-IDF v3.0-dev-495-g3a1de7db

### Hardware
ESP Wroom 32
8<------------------------ END BUG REPORT -------------------------------------

LDmitryL, 2017-09-07:
@devsaurus This is only test code. In this code you are showing long pieces of text error occurs in the last chain. If you don't close the connection then this error will be indicated on the string with send(). And if there is a close() on it.

LDmitryL, 2017-09-12:
@jmattsson Could you help me in solving this problem ?! I can't understand what's wrong and what's causing the panics? I have a problem with the closure of the chain connection notice in the previous hotfix NodeMCU. But there it was able to circumvent other algorithmic solution. On the latest version of NodeMCU this trick does not work. Wrote a simple test case I did something wrong, showed the presence of these problems.

jmattsson, 2017-09-12:
I'd love to handball this one to @devsaurus since he's the expert on `net` at the moment. Pending that, I'll see if I can find some time to investigate for myself, but I'm still really busy with Other Stuff(tm).

LDmitryL, 2017-09-12:
@jmattsson Thank you for responding to my request. 
Can add that the previous correction of this error was already present. I was able to get around remembering the `CON `as a global variable. And the connection is not closed as `c:close()` and how `global_con:close()`. And it helped. But not now. Maybe the problem is in the `close()` statement? It takes to close a wrong variable?
Managed to reproduce the error? Maybe it's a peculiarity of my version ESP32 !?

devsaurus, 2017-09-12:
Thanks for reporting @LDmitryL!
Should be fixed with b2f09da2e2f5acbc8c0c9ff97553b36b112580e6.

LDmitryL, 2017-09-13:
@devsaurus Thanks, error when closing connection was gone. But now an error occurred while transferring large files. To correct this error was not! Error `"out of memory"` statement of `send()` in `function SFC()`. 

> function SFC(SS) 
> local line=file.read(1000); 
> if line then SS:send(line, function (c) SFC(c) end); else file.close(); SS:close(); end 
> end 
> S:on("receive", function(con,reg) 
> line='\nTEST LINE\n'; file.open("style.min.css", "r"); 
> con:send(line, function (c) SFC(c); end);
>end) 
> 

To repeat this error in a short example does not work. Only when transferring files to the browser. Moreover, the HTML and CSS files are transferred without errors but in the third JS file in half causing the error. What can you recommend? When this `node.heap() = 141104`

devsaurus, 2017-09-14:
Please open a new issue since this topic is unrelated to the close bug. We'd like to keep the discussions separated.
Said that, in the issue description please also provide a hint about file/transfer sizes that trigger the error and whether the 3 files are sent in parallel or sequentially.

closed 2017-09-14 by devsaurus