nodemcu/nodemcu-firmware #2101: 2.1.0-master_20170824 app/Makefile is missing sjson and sjson library

TerryE, 2017-09-11:
### Expected behavior
The master build should make if json is enabled.
### Actual behavior
The link fails with undefined references to `jsonsl_feed` et al.  

### NodeMCU version
2.1.0-master_20170824  release

### Hardware
N/A

I just added `sjson` to the `SUBDIRS` and `sjson/libsjson.a` to `COMPONENTS_eagle.app.v6` in `app/Makefile`.

pjsg, 2017-09-14:
This is strange. When I checkout that version, I see sjson in the SUBDIRS in app/Makefile. 

See https://github.com/nodemcu/nodemcu-firmware/blob/master/app/Makefile for the current master version.

TerryE, 2017-09-14:
Oops 😣. This was me being dumb. I'll check later, but I probably copied across an earlier make and forgot.

closed 2017-09-14 by TerryE

pjsg, 2017-09-14:
Yeah -- that makes sense. I have a set of changes to the makefile as well that I keep having to be careful (a) not to lose; and (b) not to commit!