nodemcu/nodemcu-firmware #2102: PANIC: unprotected error in call to Lua API not connected

StefanL38, 2017-09-13:
The testcode below runs for 300-1500 loops and then the Lua-script crashes with panic.

to me it seems like  wifi.sta.status()  is set to wifi.STA_GOTIP before the board is really ready to execute a send-command or maybe the status "connected" gets lost again very quick. Maybe the testcode is a somehow "unlucky" mix of code-pieces. I have taken some code from the web and modified it to my needs. Anyway if a code runs for several hundred loops and then crashes there is something wrong in a manner that is unpredictable. Code should be as predictable as possible.  

I would expect the send-command not to crash but to give back a result-value "was not able to send"

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
sending data over WiFi with a connection established with
cl=net.createConnection(net.TCP, 0)
cl:connect(80,"192.168.4.1")

sending the data with the command
cl:send(MyIDString.."Hello World! ")
is only executed when 
if wifi.sta.status() == wifi.STA_GOTIP then
  cl:send(MyIDString.."Hello World! ")

### Actual behavior
if-condition delivers value wifi.sta.status() == wifi.STA_GOTIP
but Lua-Script crashes with PANIC: unprotected error in call to Lua API not connected 

### Test code
```Lua
-- 
MyIDString = " ESP8266_Client4.lua "
station_cfg={}
station_cfg.ssid="test"
station_cfg.pwd="12345678"

print(MyIDString)
wifi.sta.disconnect()
wifi.setmode(wifi.STATION) 
wifi.sta.config(station_cfg)
wifi.sta.connect() 
print("Looking for a connection")

tmr.alarm(1, 2000, 1, function()
     if(wifi.sta.getip()~=nil) then
          tmr.stop(1)
          ConnectErrorCnt = 0
          LoopCnt = 0
          print("Connected!")
          print("Client IP Address:",wifi.sta.getip())
          cl=net.createConnection(net.TCP, 0)
          cl:connect(80,"192.168.4.1")
          cl:on("receive", function(cl, pl) print(pl) end)
          tmr.alarm(2, 1000, 1, function() 
              WiFiStatus = wifi.sta.status()
              print("ConnErrCnt:"..ConnectErrorCnt)
              print(MyIDString.."direct before sended Hello World to server 192.168.4.1")  
              if wifi.sta.status() == wifi.STA_GOTIP then
                print("wifi.sta.status()"..wifi.sta.status())  
                cl:send(MyIDString.."Hello World! wifi.sta.status()"..wifi.sta.status())
                LoopCnt = LoopCnt + 1
                print(MyIDString.."direct after sended Hello World LoopCn:t"..LoopCnt)  
              else  
                ConnectErrorCnt = ConnectErrorCnt + 1
                print(MyIDString.."WiFiStatus <> wifi.STA_GOTIP: "..WiFiStatus)
                print("wifi.STA_IDLE="..wifi.STA_IDLE)
                print("wifi.STA_CONNECTING="..wifi.STA_CONNECTING)
                print("wifi.STA_WRONGPWD="..wifi.STA_WRONGPWD)
                print("wifi.STA_APNOTFOUND="..wifi.STA_APNOTFOUND)
                print("wifi.STA_FAIL="..wifi.STA_FAIL)
                print("wifi.STA_GOTIP="..wifi.STA_GOTIP)
              end  
          end)
      else
         print("Connecting...")
      end
end)

another nodeMCU-board WeMos D1 mini is running this code
wifi.eventmon.register(wifi.eventmon.STA_CONNECTED, function(T)
 print("\n\tSTA - CONNECTED".."\n\tSSID: "..T.SSID.."\n\tBSSID: "..
 T.BSSID.."\n\tChannel: "..T.channel)
 end)

 wifi.eventmon.register(wifi.eventmon.STA_DISCONNECTED, function(T)
 print("\n\tSTA - DISCONNECTED".."\n\tSSID: "..T.SSID.."\n\tBSSID: "..
 T.BSSID.."\n\treason: "..T.reason)
 end)

 wifi.eventmon.register(wifi.eventmon.STA_AUTHMODE_CHANGE, function(T)
 print("\n\tSTA - AUTHMODE CHANGE".."\n\told_auth_mode: "..
 T.old_auth_mode.."\n\tnew_auth_mode: "..T.new_auth_mode)
 end)

 wifi.eventmon.register(wifi.eventmon.STA_GOT_IP, function(T)
 print("\n\tSTA - GOT IP".."\n\tStation IP: "..T.IP.."\n\tSubnet mask: "..
 T.netmask.."\n\tGateway IP: "..T.gateway)
 end)

 wifi.eventmon.register(wifi.eventmon.STA_DHCP_TIMEOUT, function()
 print("\n\tSTA - DHCP TIMEOUT")
 end)

 wifi.eventmon.register(wifi.eventmon.AP_STACONNECTED, function(T)
 print("\n\tAP - STATION CONNECTED".."\n\tMAC of client: "..T.MAC.."\n\tAID: "..T.AID)
 table={}
   table=wifi.ap.getclient()
   for mac,ip in pairs(table) do
     print(mac,ip)
   end              
 
 end)

 wifi.eventmon.register(wifi.eventmon.AP_STADISCONNECTED, function(T)
 print("\n\tAP - STATION DISCONNECTED".."\n\tMAC: "..T.MAC.."\n\tAID: "..T.AID)
 end)

MyIDString = " ESP8266_Server3.lua "
 
print(MyIDString)
wifi.setmode(wifi.STATIONAP);
wifi.ap.config({ssid="test",pwd="12345678"});
print("Server IP Address:",wifi.ap.getip())

sv = net.createServer(net.TCP) 
sv:listen(80, function(conn)
    conn:on("receive", function(conn, receivedData) 
        print(MyIDString.."Rcv Data:#" .. receivedData.."#")         
    end) 

    tmr.alarm(1, 10000, 1, function()
        if wifi.sta.getip()== nil then
            print("Waiting for IP...")
        else
            WiFiStatus = wifi.sta.status()
            if WiFiStatus == wifi.STA_GOTIP then
              table={}
              table=wifi.ap.getclient()
              for mac,ip in pairs(table) do
                print(mac,ip)
              end              
              print(MyIDString.."send Msg Hi to client")
              conn:send(MyIDString.."Server to Client Hi Client") 
              print(MyIDString.."Msg Hi sended to client")
              table={}
              table=wifi.ap.getclient()
              for mac,ip in pairs(table) do
                print(mac,ip)
              end              
            else  
              print(MyIDString.."Wifi-Status: "..WiFiStatus)
            end  
        end
end)

end)

### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: master
	commit: 2e67ff5a639a13260fd4fb3c3b627ccdc2845616
	SSL: false
	modules: adc,bit,dht,file,gpio,http,i2c,net,node,ow,spi,tmr,uart,websocket,wifi
 build 	built on: 2017-09-13 12:06
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
### Hardware
Amica nodeMCU-board running the "client-code" WeMos D1 mini running the Servercode

8<------------------------ END BUG REPORT -------------------------------------

TerryE, 2017-09-14:
@StefanL38, IMO, this isn't a bugrep; it's a "why doesn't my app work as I expected?"
A `sck:send()` will fail is the socket is not connected, or has been disconnected, or the previous send hasn't been sent.  That's why we have the associated on callbacks.  So at the statement after  `cl:connect(80, "192.168.4.1")`  the socket isn't connected and there is no guarantee that the socket will be connected 2sec later or will ever connect because that requires a handshake between the client and the server.

Use the proper events to sequence your code.  Don't make arbitrary timing assumptions, because these may occasionally be wrong.   If an API call throws and your code doesn't catch it, e.g. by wrapping the routine in a pcall to get the error traceback then you will get a panic.  

This example isn't a workable basis of a bugrep. Sorry.

closed 2017-09-14 by TerryE

StefanL38, 2017-09-14:
Hello Terry,

thank you very much for answering. 
I would be even more thankful if you could post a link to a proper coded example.

I went through all the Lua-commands to find things like error-handling or how to check if a connection is established. I did not find more than I used in my code. I have read a little-bit about pcalls but to me it seems that I have to add my own lua-modules written in C to make use of pcalls.

If I have established a TCP-connection and this connection was running for 5 minutes how would I do handshaking between client and server in a proper way to avoid getting thrown out by a panic-error who says "not connected"?

I'm no TCP/IP-specialist but is TCP not meant to do a lot of checking inside the TCP-protocol?

It would be very helpful for me If you could post a link to a website that shows how it has to be done the proper-way. I did not find it myself even after hours of googling.

best regards  

Stefan