nodemcu/nodemcu-firmware #2105: ESP32: Error "out of memory" when sending a large buffer

LDmitryL, 2017-09-14:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Sending data to the client.
### Actual behavior
The error `out of memory` when transferring large file in blocks of 1000 bytes.
The memory size at the time of the error node.heap()=142740
The size of transmitted data (where the error occurs) from about 40 to 80Kb. Probably depends on the speed of transmission. Prior to the occurrence of error is seen slowing packet transmission operator 'send()'. This error appeared after hotfix #2096 `Bug in net.server:close()`.

### Test code
```Lua
function SFC(SS)
     local line=file.read(1000);
     if line then 
          SS:send(line, function (c) SFC(c) end); 
     else 
          file.close(); 
          SS:close(); 
     end
end

S:on("receive", function(con,reg)
       line='\nTEST LINE\n'; 
       file.open("LargeFile.bin", "r");
       con:send(line, function (c) SFC(c); end);
end)
```

### NodeMCU version
NodeMCU ESP32 build unspecified powered by Lua 5.1.4
ESP-IDF v3.0-dev-495-g3a1de7db

### Hardware
ESP Wroom 32

8<------------------------ END BUG REPORT -------------------------------------

LDmitryL, 2017-09-14:
@devsaurus Write a short sample program failed. The test program 52 line works without error. But memory was more. `heap=225872.`

LDmitryL, 2017-09-15:
@devsaurus In search of the causes of errors I got on such cases. (I don't know, maybe someone will help): Sometimes when transferring large files (pass 3 pieces at 45kB approx) speed when the second file starts to fall, then stops. And forth from the situation or get in the console the message: 
>I (75870) wifi: bcn_timout,ap_probe_send_start 
>I (76370) wifi: bcn_timout,ap_probe_send_start 
>I (76870) wifi: bcn_timout,ap_probe_send_start 
>I (77370) wifi: bcn_timout,ap_probe_send_start 
>I (77870) wifi: bcn_timout,ap_probe_send_start 
>I (78380) wifi: bcn_timout,ap_probe_send_start
 
 And so on to infinity. while is patience.
Or get the error `"out of memory"` in function `SS.send()`
This is not evident in simple examples. Only if you have a lot of code in other places, not only in the communication part. While the transfer function is time-consuming and sending big buffer in one call! I pass on 1000b and chain call length of 45 times. And so maybe 3 times in a row.

Do not use in the code to handle events `listen() `
`S:on("receive", function (con,reg) receiver(con,reg); end) `
the function definition receiver as 
`receiver=function (con,reg) -- to do; end `
Though it is valid in LUA! This, under mysterious circumstances, sometimes leads to errors 
`PANIC: unprotected error in call to Lua API (attempt to call a nil value)`
Write better: `function receiver() -- to do; end`. Preferably without a local function. 
Helped me to reduce the `PANIC...`

LDmitryL, 2017-09-15:
Comments for writing of the WWW server. Do not drive function in the receive event handling for 
`GET={};`
...
`S:on("receive", function (con,reg)  --ToDo con & reg;  end) `
All modern browsers are trying to make multiple connections at once. Better to remember them in the table and then processing. 
`S:on("receive", function (con,reg) GET[#GET + 1] ={con,reg}; receiver(); end)` 
This will prevent your code from many of the problems associated with the feature of communication ESP32

devsaurus, 2017-09-15:
I tried to reproduce this issue with a very basic example. On ESP32:

```lua
S:on("receive", function(con,reg)
       line='\nTEST LINE\n'; 
       file.open("LargeFile.bin", "r");
       con:send(line, function (c) SFC(c); end);
end)
```

On client side:
```
nc ip_of_esp32 8080 > /dev/null
```
After typing a random line to nc it takes 3-4 seconds to transfer the file (114500&nbsp;bytes). A lot of loops without error so far.

devsaurus, 2017-09-15:
> All modern browsers are trying to make multiple connections at once.

If you run serve the file to such a browser with the original example of my test case, then you will face an issue with concurrent file access. Only one file can be open at a time: starting a second transfer breaks file reading for the first.
Apart from this, there *should* be no restriction to have 2 or more sockets sending in parallel.

LDmitryL, 2017-09-18:
@devsaurus The problem with the error "out of memory" was partly solved by reducing the length of data to be transferred in a single call to send() with 1000b to 512b. But due to the time of transmission. Now transfer 81,4 kB is 43,78 seconds. And was 23. Restrictions on transfer socket no. All incoming I accept and remember. And waiting for completion of transfer of the previous file start sending the next. The problem is probably the fact that in addition to the server programs there are other tasks running on timers. And sometimes they apparently face in time. Therefore, it is impossible to reproduce the error just by running the example. Everything works normally without problems. But when I artificially increase the speed of query errors occur in transmission.

closed 2017-09-20 by LDmitryL