nodemcu/nodemcu-firmware #2109: mqtt.client:on('connect') event does not fire if also using callback in mqtt.client:connect()

oyooyo, 2017-09-17:
8<------------------------ BUG REPORT -----------------------------------------
A callback registered for the mqtt.client:on('connect', callback) event will not be called if also using the callback in the mqtt.client:connect() function call. The example code in the mqtt module suggests otherwise.

### Expected behavior
The callback registered via mqtt.client:on('connect') should be called (preferred solution), or the documentation should at least be updated to mention this conflict.

### Actual behavior
The callback registered via mqtt.client:on('connect') won't get called, but the example code in the mqtt module docs suggests that it will.

### Test code
Taken from the example code in the mqtt module docs:
```Lua
m = mqtt.Client("clientid", 120)
m:on("connect", function(client) print ("connected_1") end)
m:connect("192.168.11.118", 1883, 0, function(client)
  print("connected_2")
end)
```
Only "connected_2" gets printed, but not "connected_1".

### NodeMCU version
Latest dev

### Hardware
Standard NodeMCU-DevKit

8<------------------------ END BUG REPORT -------------------------------------

oyooyo, 2017-09-17:
I just had a short look at the source code of the mqtt module, which led me to believe that the issue will probably be "solved" by updating the documentation.

Would it be possible to instead add a way to query/poll the MQTT client for its current connection status (connected/offline)? Something like for example mqtt_client.is_connected or mqtt_client:is_connected() ?

marcelstoer, 2017-09-17:
While the behavior is quite logical to me (i.e. it makes sense and is expected) I agree that https://nodemcu.readthedocs.io/en/latest/en/modules/mqtt/#example is far from ideal. Could you suggest a new structure for `mqtt.md` in a PR?

oyooyo, 2017-09-18:
The behavior makes sense if one is aware that

a) NodeMCU (always?) only allows one event listener per event, and registering for an event the second time will silently overwrite the first event listener reference. But for people who are already familiar with Javascript/Node.js, this behavior is not what one would intuitively expect. I assume this behavior is explained somewhere in the NodeMCU docs, but I couldn't even instantly find it.

b) the callback provided in the mqtt.client:connect() function call is simply being registered as a listener for the 'connect' event, and is basically just a convenience that will save an additional mqtt.client:on('connect',...) call. Since one can also provide an additional callback for the failure case in the mqtt.client:connect() call, I instead expected the two callback arguments of the mqtt.client:connect() function to act like success/failure callbacks.

Anyway, what about introducing a way to poll the current mqtt connection status? Something similar to wifi.sta.status() ?

closed 2019-08-04 by stale[bot]