nodemcu/nodemcu-firmware #2111: get request with basic authentication

chathurangawijetunge, 2017-09-19:
Is it possible to send GET request with basic authentication

Eg:- username:password@xxx.com

As when I use ':' it assumes as the port.

marcoskirsch, 2017-09-19:
[nodemcu-httpserver](https://github.com/marcoskirsch/nodemcu-httpserver) supports that. You can look at how it does it or use the project outright.

marcelstoer, 2017-09-19:
Sure, basic auth is nothing but a base64 encoded "user:password" value in the HTTP `Authentication` header. Hint: `encoder.toBase64()`.
Also see #1010.

closed 2017-09-19 by marcelstoer

chathurangawijetunge, 2018-03-05:
it dose not work
Eg.-

http://username:password@dynupdate.no-ip.com/nic/update?hostname=mytest.testdomain.com&myip=1.2.3.4

Base64 encode of "username:password" == dXNlcm5hbWU6cGFzc3dvcmQ=

so if i do
http://dXNlcm5hbWU6cGFzc3dvcmQ=@dynupdate.no-ip.com/nic/update?hostname=mytest.testdomain.com&myip=1.2.3.4

shouldn't it work ...? but it is not  i get

> HTTP client: DNS failed for xxxxxxxxxxxxxxxxxxxx@dynupdate.no-ip.com

what is the correct way?