nodemcu/nodemcu-firmware #2113: how should i send hex data use nodemcu?

ctlPython, 2017-09-20:
I want to send a bunch of hex data to my sensor use NodeMCU.What should i do?

curlyz, 2017-09-20:
with uart or i2c , I learned the hard way that you cant send a string directly , use a for loop to send each bytes.

>>setup uart 
>> for i = 1 , sth do uart.write(...., data[i] )
>>watch out for nil

ctlPython, 2017-09-20:
thank you.....urat.write() function is send a string .But the sensor must send a hex data that it will return some data to me. i don't know what should i do . can you teachme that?

curlyz, 2017-09-20:
Just for reference , uart.write(0,string.byte(data[i])) will send hex , use this in a loop.
Secondly , if you need to receive data , it will be tricky 
Use uart.on  , if your protocol is with a fixed string length then use uart.on with callback on bytes received that you have known  
If not a fixed length then set callback on bytes received to 1 and do function inside to accumulate the data as you wish.
May I ask what device are you working with ?

ctlPython, 2017-09-20:
ZPHS01 

i should send 11 02 01 00 EC hex data to the sensor. And i will get a response to receive 14 bits data.

ctlPython, 2017-09-20:
@curlyz like this? but it doesn't work

`uart.setup(0, 9600, 8, uart.PARITY_NONE, uart.STOPBITS_1, 1)

print(uart.getconfig(0))

data = {"11", "02", "01", "00", "EC"} 
for i = 1 , 5 
do  
    print(data[i])  
    uart.write(0,string.byte(data[i]))
end  

uart.on("data", 14,
  function(data)
    print("receive from uart:", data)
    if data=="quit" then
      uart.on("data") -- unregister callback function
    end
end, 0)`

curlyz, 2017-09-20:
data = {"11", "02", "01", "00", "EC"}   >>> add 0x to become hex data
U are talking about 14 bits not bytes ?? wew :))

ctlPython, 2017-09-20:
thank you very much. i use your method but it doesn't work. I noticed that the data it printed was 0x11 00x02 00x01 etc...

curlyz, 2017-09-21:
```
q3k.I2C.Write = function(Address, ...)
	print("I2C Write")
    local Data = {...}
    local Bytes = string.format("%02x ", Address)
    for _, Byte in pairs(Data) do
        Bytes = Bytes .. string.format("%02x ", Byte)
    end
    local DataString = string.char(unpack(Data))
    return i2c.write(q3k.Config.I2CBus, Address, DataString)
end
```
You can do this the same with uart. I found this wonderfull sketch online.

ctlPython, 2017-09-21:
okay, i will try it. thank you very much~

ctlPython, 2017-09-23:
@curlyz I have successfully solved my problem. Now I get fourteen hex data,but ESPlorer print some messy code. i use uart assistant can see the hex data. how should i get the data use my nodemcu?

TerryE, 2017-09-23:
Guys, this sort of discussion doesn't belong on this issues list.  Sorry.

closed 2017-09-23 by TerryE

curlyz, 2017-09-24:
i2c.start(id)
     i2c.address(id, dev_addr, i2c.RECEIVER)
     str = i2c.read(id, 8)
     i2c.stop(0)

     b = {}
     str:gsub(".",function(c) table.insert(b,c) end)

     if string.byte(b[1]) ~= 3 then return nil end --checking for correct response
     if string.byte(b[2]) ~= 4 then return nil end --number of bytes returned, should be 4
 Use this man and close the issue