nodemcu/nodemcu-firmware #2114: Module for BME680

Necromancer1982, 2017-09-21:
### Missing feature
Module for BME680 is missing

### Justification
The new [BME680 sensor ](https://www.bosch-sensortec.com/bst/products/all_products/bme680)from bosch is the follow-up chip of the already supportet BME280.
Beside temperature, humidity and air pressure, **AIR QUALITY** (VOCs) could be messured.
In my opinion the BME680 is a cheap solution to messure IAQ-Index (Indoor Air Quality) for IoT and home automatisation solutions like automatic aeration.

### Workarounds
The only workaround is to directly communicate via I2C to the Chip.

marcelstoer, 2017-09-21:
Don't know if maybe @vsky279 (implemented the BME280) has time, interest and necessary hardware to implement this. Otherwise it _may_ get closed as per #1010.

vsky279, 2017-09-21:
I can't find BME680 on ebay.
Bosch pusblishes the driver (https://github.com/BoschSensortec/BME680_driver) so it should not be that difficult to implement it. Maybe by extending directly the BME280 module.
I would by one if I found it and would try to implement it.

marcelstoer, 2017-09-21:
Seems to be difficult to source. Not on ebay, seeedstudio, bangggod or aliexpress.

There are some (expensive) breakouts at http://www.watterott.com/en/BME680-Breakout and https://oshpark.com/shared_projects/LXuNziGd and Mouser started selling the sensors at http://eu.mouser.com/ProductDetail/Bosch/BME680.
This post http://arduino-praxis.ch/2017/09/08/bme680-umweltsensor-board/ suggest that there's another BOB in the making.

Maybe @Necromancer1982 can have a BME680 shipped to your door as an exchange for you to write the module 😜

Necromancer1982, 2017-09-22:
Hey guys,

I've wrote a comment on the blog of http://arduino-praxis.ch, if they are interested in a collaboration with you, to give you a sample for developing the module. I'm curious to hear from them.

awatterott, 2017-09-30:
If there is still interest in a BME680 breakout then we can offer one for implementation.

vsky279, 2017-09-30:
Hi Andreas @awatterott, yes, we are still interested. It would be nice to support it in @nodemcu.

awatterott, 2017-09-30:
Please write us a short mail with your postal address: support (at) watterott.com

vsky279, 2017-10-07:
I confirm reception. Now to find time to play with it. Thank you!

vsky279, 2017-10-17:
Hi,
I started to work on the driver. What is clear at this moment is that a separate driver from the BME280 needs to be programmed. It would be very complicated to implement BME680 into existing driver and would not make much sense.

But I started from the existing BME280 driver and trying to adjust it to the BME680. I'm using `bme680_defs.h` as provided by Bosh.
So far I can talk to the sensor and read temperature, pressure and humidity but at this moment the values don't make sense.
I have not touched the air quality sensor readout yet.

You can check the work-in-progress version here:
https://github.com/vsky279/nodemcu-firmware/blob/BME680/app/modules/bme680.c

vsky279, 2017-10-28:
I have created a PR #2151 with BME680 module.

marcelstoer, 2017-11-19:
@awatterott thanks for the board.

closed 2017-11-19 by marcelstoer

awatterott, 2017-11-19:
For open-source things there is in general a support from us possible. If there is a need in the future, just write us a short mail.

PR #2151 (2017-11-19): BME680 module initial commit
Fixes #2114.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

This is a simple driver of the Bosch's BME680 sensor. In ad [... cut]