nodemcu/nodemcu-firmware #2116: MQTT lwt not working

dbaumgarten, 2017-09-23:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

The lwt() function of MQTT should set up the clients last will on the MQTT-Broker.

### Actual behavior

The lwt() function does nothing. A tcpdump (tcpdump -n port 1883)  on the broker reveals, that no package is received by the broker when running the lwt() function.

The broker is mosquitto 1.4.14.

### Test code
When connected to the broker and running:
```mqttclient:publish("node/heater1/status","bla",2,1)```
the message is properly published and tcpdump shows network-activity.
But when running:
```mqttclient:lwt("node/heater1/status","off",2,1)```
tcpdump shows no network-traffic and after restarting the nodemcu-board no message is published to the topic.

### NodeMCU version
Current master head.

### Hardware
Nodemcu v2 dev-board

8<------------------------ END BUG REPORT -------------------------------------

dbaumgarten, 2017-09-23:
I just had a look at the MQTT-Specification (I should have done this far earlier) and it turns out, that the last will is transmitted when the connection is initiated. This means lwt() must be called before connect().
Also the MQTT-Broker (at  least mosquitto) will wait for the duration of the timeout specified in connect() before it publishes the last will of the client.

Tl;dr: The library works correctly, but the documentation could be more clear on how to use lwt(). I would open a PR for this, if it's ok.

marcelstoer, 2017-09-28:
> the documentation could be more clear on how to use lwt(). I would open a PR for this

Great, looking forward to this. I stated elsewhere that I find the example at https://github.com/nodemcu/nodemcu-firmware/blob/dev/docs/en/modules/mqtt.md#example really not very helpful as it suggest a certain sequence of function calls that is not ideal.

closed 2017-10-05 by marcelstoer

medilies, 2020-07-17:
hello @dbaumgarten 
Can you show me a little example of using LWT in nodeMCU and which library should i include

dbaumgarten, 2020-07-18:
Sure, here is how I am using it:
```
local config = {}
config.broker = "brokerip"
config.port = 1883
config.user = "username"
config.pwd = "passwword"
config.name = "node1"

 m = mqtt.Client(node.chipid(), 60, config.user, config.pwd)
    m:lwt("node/"..config.name.."/status","OFFLINE",2,1)
    m:on("message", functiontocallwhenmessagereceived)
    m:on("offline",functiontocallwhenconnectionlost);
    m:connect(config.broker, config.port, 0, 0, function(con)
        m:publish("node/"..config.name.."/status","ONLINE",2,1)
        print("Connected to MQTT-Broker!")
        M.onconnect(con)
    end, mqqt_on_error)
```

You will of course have to do the whole wifi connection stuff first.

PR #2128 (2017-10-05): Better documentation for mqtt:lwt()
Fixes #2116 
Like #2125 but this time for dev.