nodemcu/nodemcu-firmware #2121: undefined reference to `__ctype_ptr__'

scJingP, 2017-09-29:
lua/.output/eagle/debug/lib/liblua.a(lobject.o): In function `luaO_rawequalObj':
/cygdrive/d/Workspace_C/nodemcu-firmware-master/app/lua/lobject.c:95: **undefined reference to `__ctype_ptr__'**
collect2: error: ld returned 1 exit status
make: *** [../Makefile:340: .output/eagle/debug/image/eagle.app.v6.out] Error 1
现在无法编译，请问这个问题该怎么解决？

marcelstoer, 2017-10-01:
> `/cygdrive/d/`

I doubt that building under cygwin is supported.

djphoenix, 2017-10-02:
@marcelstoer yep, it's supported with Espressif' crosstool fork.
@scJingP which version of toolchain you're use?

TerryE, 2017-10-20:
I have closed this because it is over 3 weeks since the OP raised the issue -- probably because the issue has been solved.  We can always reopen it if needed.

closed 2017-10-20 by TerryE