nodemcu/nodemcu-firmware #2123: EUS not connecting to WiFi on second start

skycoders, 2017-10-01:
### Expected behavior
After providing WiFi credentials for EUS, on the next boot it should try to connect to the WiFi and close EUS if successful.

### Actual behavior
In DEV branch, EUS does not connect to WiFi, even though the credentials are stored. It seems that since the new WiFi config table has been introduced and the auto connect flag must be set explicitly, this is not done by EUS. At least I could not find anything indicating this in the code.
If I call wifi.sta.connect() manually, EUS finishes as expected after the connection has been established. After writing the configuration again with auto=true, it also behaves as expected, connecting automatically and closing EUS.

### Test code
```Lua

enduser_setup.start(
      function()
        -- on first time setting credentials, this gets called. Afterwards never again
         print("connected")   
     end,
      function(err, str)
        print("enduser_setup: Err #" .. err .. ": " .. str)
      end
    );

```
### NodeMCU version
DEV branch, rev ce1712d1

### Hardware
Sonoff S20 smart plug, Sonoff Smart Switch

marcelstoer, 2017-10-01:
> auto connect flag must be set explicitly

Hhmm, where did you get that from? [auto connect is on](http://nodemcu.readthedocs.io/en/latest/en/modules/wifi/#syntax_14) by default.

skycoders, 2017-10-02:
I know, it used to be like this. I also found the code in the wifi module where auto connect should be set, however with flashing a newly built firmware from today (rev 3e60fa8f) it still does not connect the WiFi again after using EUS. I currently have no clue why this happens, but it does, also with more than one device.

marcelstoer, 2017-10-02:
Just to avoid any misunderstandings, I wasn't suggesting that this must be a problem on your end, sorry. Only wanted to confirm that auto connect is or should be on by default.

@dnc40085 would you have time to confirm that all is well on the WiFi side?

skycoders, 2017-10-02:
all fine :) no problem. I just found out that when I load the config which got stored by EUS, and then set auto=true and set it again with wifi.sta.config(), it works from then on. 

I tried something else: If I create a wifi config with only ssid and pwd and use wifi.sta.config() to set it, I get connected automatically, and also after a reboot, autoconnect works. However, clearing the config and doing the same with the EUS module results again in not auto connecting. I couldn't find a hint in the code how this can happen so far, but something seems to be strange.

skycoders, 2017-10-02:
what I did find now: the wifi module calls wifi_station_set_auto_connect() out of the OS SDK, EUS does not. This call is also performed by the wifi module if no "auto" parameter is given, which is the known  behavior. My impression is that this call needs to be added to the do_station_cfg() call of EUS.

closed 2019-08-04 by stale[bot]