nodemcu/nodemcu-firmware #2127: When enduser_setup.html is edited, makefile should call prepare.sh

Jcrash29, 2017-10-04:
This should be a pretty easy makefile change to detect enduser_setup.html changes and declare a call to prepare.sh

I'm not great with makefiles.... So I'm making the Issue so I can learn how to fix this simple issue.

marcelstoer, 2018-02-24:
Maybe you can fix this when you complete the pending PR?

TerryE, 2018-03-01:
No.  This is the wrong question.  @robertfoss oss, @jfollas the real question is why does EUS not follow the conventions of the rest of NodeMCU, or at least not implement a convention that could be extended to other modules?

- It has a subdirectory under modules for what is essentially source code. 
- It uses a shell script in this subdirectory for doing the compile and as @Jcrash29 observes, this isn't hooked into the make chain, but is used for compiling a .html file into a C header file in a non-standard way
-  It requires the installation of a non-standard version of gzip (zopfli) to save 76 bytes of out of 2.5K -- just not worth it.

In principle I feel that we should "just call a spade a spade" and move the html file into the `modules` directory and get rid of `eus`.  We should also add a generic rule for `%.h: %.html` which does this transformation generically and modify the end user module accordingly so people that want to use this code and also want to modify the html file can just do so and remark, letting make what it was designed to do.

I personally don't like this module, but I have to accept that it is popular amongst our users, so this should be done properly.  I'll put my name on this one and slot it in after my next big PR release, saying as I am trying to sort out other make anomalies at the same time.  

My suggestion

closed 2019-07-21 by marcelstoer