nodemcu/nodemcu-firmware #2130: Rewrite of rfswitch module to use the hardware timer.

jonlidgard, 2017-10-10:
8<------------------------ FEATURE REQUEST ------------------------------------
The rfswitch module relies on the far less accurate os_delay_us to implement the pulse train.
I've implemented a hw_timer interrupt based version which should be more reliable.
The interface is identical to the original so shouldn't break anything unless you are also using the pwm features on the esp8266 which is mutually exclusive with the use of hw_timer.

### Justification
Increased reliability, more responsive.

### Workarounds
[https://github.com/jonlidgard/nodemcu-firmware/blob/master/app/modules/rfswitch.c](url)

Willing to add it as a Pull Request if asked.
8<------------------------ END FEATURE REQUEST ------------------------------

marcelstoer, 2017-10-19:
Feel free to make a PR so @ffedoroff (original author of the module) can review and comment on it.

ffedoroff, 2017-10-19:
@marcelstoer yes please send PR and I will work on it. Thank you!

marcelstoer, 2018-02-24:
> Willing to add it as a Pull Request if asked.

@jonlidgard can we take you up on this? Will otherwise close this.

closed 2018-07-20 by marcelstoer