nodemcu/nodemcu-firmware #2138: KRACK vulnerability

pjsg, 2017-10-17:
I see that Espressif has announced that they have resolved this issue -- so we should probably do a release with the new SDK:

http://espressif.com/en/media_overview/news/espressif-releases-patches-wifi-vulnerabilities-cert-vu228519#.WeVJ91XhAqE.twitter

heythisisnate, 2017-10-17:
Thank you for raising the issue. I came here looking for a discussion on this vulnerability.

marcelstoer, 2017-10-17:
Waiting for https://github.com/espressif/ESP8266_NONOS_SDK/issues/46

marcelstoer, 2018-01-10:
If I interpret @FayeY's and @wujiangang's comments on espressif/ESP8266_NONOS_SDK#46 correctly then there won't ever be an official 2.1.x release with just the KRACK fix but a 2.2 later this month.

marcelstoer, 2018-02-06:
Superseded by #2225.

closed 2018-02-06 by marcelstoer