nodemcu/nodemcu-firmware #2139: ESP32: the transmission speed of the network?

LDmitryL, 2017-10-18:
Decided to test transfer speed. And got very strange results! 
If the packet is long: 
242Kb load time is 2.13 seconds. 
194Kb passed of 3.04 seconds. 
97.4 Kb is transmitted 21.81 seconds. 
If you reduce the packet size to 

> line='['..tostring(number)..'] '..string.rep("#",10)..'!\n'

 and put `number=1000;` We get the transfer speed is 17.5 Kb and the download time 4.3 minutes
It turns out the shorter the package, the slower the speed? What could be the problem?
Speed was measured by the Chrome browser in debug mode page.

### Test code
```Lua
station_cfg={};
wifi.start();
wifi.mode(wifi.STATION);
station_cfg.ssid ='XXX.XXX.XXX.XXX' 
station_cfg.pwd ='123456789'  
wifi.sta.config(station_cfg,false);
number=0;

function SFC(SS)
    number=number-1;
-- Change the size of the package 97,4Kb .. 194Kb..242Kb
    line='['..tostring(number)..'] '..string.rep("#",1000)..string.rep("&",1000)..string.rep("*",500)..'!\n';
    if number<1 then line=nil; end;
    if line then  SS:send(line, function (c) SFC(c) end); else print("Close"); SS:close();  end
end

srv=net.createServer(net.TCP,30);
srv:listen(80, function(S)
          S:on("receive", function(con,reg) number=100;  line="begin\n"; con:send(line, function (c) SFC(c); end);  end); 
end);
```
### NodeMCU version
Latest dev-esp32
### Hardware
ESP-WROOM-32

LDmitryL, 2017-10-18:
@devsaurus You can comment on this behavior. It turns out that for small packets the speed of the device will be bad?

devsaurus, 2017-11-12:
Yes, I can confirm such characteristics. It appears that the overall overhead becomes dominant when you decrease the payload size and increase the repeats. I don't see a way to avoid this.

Alongside I discovered some issues with big send data. Pushed an update on `dev-esp32` to fix this.

LDmitryL, 2017-11-24:
@devsaurus  In this case, works well the classic method of sending data: 
```
conn:send("little data")
conn:send("little data")
 ... more ... 
conn:send("little data")
```
If the fill rate of the buffer is less than the speed of sending data all is well. The measured speed browser good. But if you try to pass faster than the speed of WiFi when the buffer is full receive overflow error. 
Need a function which returns the remaining size of the transmit buffer in system. Then it is possible before issuing an error to reduce the size of transmitted data and switch to data transfer style: 
`conn:send("data", function sendnextdata() end)`
 Once the buffer is empty by 50-70% activate the style `conn:send("little data")` This will reduce the overhead of calling the `send()` function associated with the transmission of dynamic data.

pjsg, 2017-11-24:
You should be using the 'sent' callback to know when it is safe to send again.

Having said that, the 'sent' callback isn't quite the right callback -- it severely limits the throughput rate. It would be possible (and I had a branch with this in it) to make the 'sent' callback come much quicker -- maybe I'll dig it up and put up a PR....

TerryE, 2017-11-26:
IIRC, the SDK doesn't marshal outbound packets.  If it did, then an application would have no guarantee that "sending"  a record would actually send it immediately without some batch logic including a dwell or a change to the API that would break existing applications.  So in general it is more time and network efficient to batch up data.  So for example the example `lua_modules/http/http.lua` includes the code fragment (comments removed):
```Lua
    if self.send_header then
      c:send("HTTP/1.1 ")
      c:send(tostring(status or 200))
      c:send(" OK\r\n")
      self:send_header("Transfer-Encoding", "chunked")
    end
```
This would better coded:
```Lua
    if self.send_header then
      c:send(("HTTP/1.1 %s OK\r\n%s"):format(
        tostring(status or 200), 
        self:get_header("Transfer-Encoding", "chunked")))
    end
```
    
  
  c:send

LDmitryL, 2017-11-27:
@TerryE I currently use` "This would better coded"` style. But when loading client I pass the script file Jscript the size of 80kb. 80 call `c:send()` with data buffer size 1kb. The measured transfer time is 25 seconds with the free WiFi channel. And can reach 1-2 minutes when the network is busy. And when using `"bad style"` speed when loading no more than 5 seconds in worse case. But errors occur when a system overflow of the transmit buffer. This transfer rate is disappointing. And negates the possibility of using NodeMCU in implementations of IoT. What's sad! The Arduino is faster. But I like LUA.

LDmitryL, 2017-11-27:
@TerryE And most importantly, why the difference in transfer speed when using different styles! Can each call chain in the `"better coded"` waiting for the release of buffer in the system? But simply put the data in and out. And when the buffer is filled here expect free memory to the long size of the current data. 
I can't explain this difference in speeds!

closed 2017-12-14 by LDmitryL