nodemcu/nodemcu-firmware #2143: bad argument #1 to 'config' (config table not found!)

kingdevnl, 2017-10-19:
The example code on the Website gives me this error
print(wifi.sta.getip())
--nil
wifi.setmode(wifi.STATION)
wifi.sta.config("SSID","password")
print(wifi.sta.getip())
--192.168.18.110

I download the firmware from the Cloud build site

marcelstoer, 2017-10-20:
Another one for #1010 ... and http://nodemcu.readthedocs.io/en/master/en/modules/wifi/#wifistaconfig

closed 2017-10-20 by marcelstoer

timothyparez, 2021-12-20:
The reason people keep stumbling on this one is because the example by @kingdevnl  
is the first thing we see on the official website while the [updated docs ](https://nodemcu.readthedocs.io/en/release/modules/wifi/#wifistaconfig)tell a different story.

From the [official website](https://www.nodemcu.com/index_en.html):

```
print(wifi.sta.getip())
--nil
wifi.setmode(wifi.STATION)
wifi.sta.config("SSID","password")
print(wifi.sta.getip())
--192.168.18.110
```
And from the docs:

```
station_cfg={}
station_cfg.ssid="NODE-AABBCC"
station_cfg.pwd="password"
station_cfg.save=false
wifi.sta.config(station_cfg)
```
At some point the API must have changed and the main page was never updated.  
I ran into this issue today.  
Perhaps we can re-open this issue until the main page is fixed?