nodemcu/nodemcu-firmware #2144: rtc time not kept through deep sleep

mkresse, 2017-10-19:
### Expected behavior

Setting the rtc time using `rtctime.set(0,0)` enables the realtime clock so that after a deep sleep, a call to `rtctime.get()` returns the curren (up-) time. This works on a version build from master on 2017-06-17. 
The example code outputs "phase 1", goes to deepsleep and then prints "phase 2".

### Actual behavior

Using a current master build (2017-10-12), reading the time using `rtctime.get()` always returns 0.
The example code outputs "phase 1" in an endless loop. I suppose that the changes of commit d93465c have something to do with the changed behaviour?

### Test code
```Lua
    local sec, usec = rtctime.get()
    if (sec == 0) then
        print("phase 1, initializing time + sleeping 2s");
        rtctime.set(0, 0)

        tmr.create():alarm(100, tmr.ALARM_SINGLE, function()
            rtctime.dsleep(2 * 1000 * 1000)
        end)
        
    else
        print(string.format("phase 2, uptime: %ds", sec));
    end
```
### NodeMCU version

Working version:
````
NodeMCU custom build by frightanic.com
	branch: master
	commit: c8ac5cfb912ff206b03dd7c60ffbb2dafb83fe5e
	SSL: false
	modules: adc,bit,cron,file,gpio,mqtt,net,node,rtcmem,rtctime,sntp,tmr,uart,wifi,ws2812
 build 	built on: 2017-06-17 20:54
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
````

Broken version:
```
NodeMCU custom build by frightanic.com
	branch: master
	commit: 2e67ff5a639a13260fd4fb3c3b627ccdc2845616
	SSL: false
	modules: adc,bit,cron,ds18b20,file,gpio,mqtt,net,node,rtcmem,rtctime,sntp,tmr,uart,wifi,ws2812
 build 	built on: 2017-10-12 19:32
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)

```

### Hardware

ESP8266 ESP-12F on custom adapter board (http://www.mikrocontroller-elektronik.de/esp12e-tutorial-einstieg-mit-dem-esp8266-modul/)

marcelstoer, 2017-10-20:
> I suppose that the changes of commit d93465c have something to do with the changed behaviour?

One for @pjsg then?

pjsg, 2017-10-20:
I'll take a look -- the first thing that occurrs to me is that it is entirely possible that the RTC now doesn't count if it is set to zero (it may treat this as 'unset').

mkresse, 2017-10-20:
@pjsg Thanks for looking into it. I just tried to set it to a different value - same behaviour.

pjsg, 2017-10-20:
I'll take a look at the code....

pjsg, 2017-10-25:
Ugh. I borked it when I made those changes. I'm making progress towards fixing it. One of the challenges is that *some* of the code runs before the bss segment is initialized.

pjsg, 2017-10-25:
I have a branch at https://github.com/pjsg/nodemcu-firmware/tree/fix-dsleep which (I think) fixes the problem. I'm going to do some more testing.....

pjsg, 2017-10-25:
It appears that (currently) a 10 second deep sleep causes a 200millisecond clock skew, A 20 second deep sleep is a 600millisecond clock skew. This seems bad.... More investigation needed....

PR #2148 (2017-10-28): Fixes problems with time not preserved through dsleep
Fixes #2144.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

When I moved the rtc variables into memory (from the very slow rtc memory), the .bss initialization would clear them out after they were initialized. This caused no end of problems. Also, the sleep tracking values were updated *after* the las [... cut]
closed 2017-10-28 by marcelstoer