nodemcu/nodemcu-firmware #2154: More problems with (lack of) deep sleep

pjsg, 2017-10-29:
Copied from #2144

Unfortunately I discovered only now an issue with the updated version. When powered up (after some seconds being turned off), rtctime.get() returns a random value instead of zero. This value remains constant even after a reset but changes to a different random value after a power down.

Only when rtctime is initialized via rtctime.set(0), it behaves as the previous version: after resetting the ESP, rtctime.get() constantly returns 0 until rtctime.set(0) is called again.

Or to describe it another way: an init.lua with:

print(rtctime.get())
now prints a random value after being powered up, while formerly it reliably printed 0.

mkresse, 2017-10-31:
I let the ESP run over night with continuous 120s deepsleeps and tested some power downs and resets:  looks good and works just like expected. :thumbsup:

However one thing (that doesn't concern me though) I noticed while experimenting: if I call `rtctime.set(0)` although rtctime has already been enabled before, the timing of a following `rtctime.depsleep()` is off by a multiple of the requested sleeping time. For example an init.lua (I did it manually though) with something like :

    print(rtctime.get())
    rtctime.set(0)
    rtctime.deepsleep(3*100000)

would:
* print 0 after power up and sleep 3 seconds,
* print 3 after 1st wakeup but them sleep ~10 or more seconds
* print 10 after 2nd wakeup

Maybe this is an expected side effect of the new calibration feature that collides with the somewhat unexpected resetting of the time to 0 (if absolute time does not matter). Maybe the calibration could be omitted in this special case?

pjsg, 2017-10-31:
Let me take a look at this.... I think that you are right, we ought to ignore the setting of the time to 0 -- it isn't an accurate set operation. The code assumes that whenever you set the time, you are setting it to the right time and then uses that to figure out what the error was during the deepsleep.

pjsg, 2017-11-02:
I've changed the bahavior of setting the time to 0. It doesn't update the sleep calibration, This code

```
print(rtctime.get())
rtctime.set(0)
rtctime.dsleep(3*1000000)
```

Always prints out 3 and about quarter of a second, and then waits for about three seconds and restarts. I'm going to have to wipe the device to get control back!

The PR includes this fix.

PR #2155 (2017-11-03): Another dsleep fix (or rather a not-deep-sleep case)
Fixes #2154.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

We now always reset the configuration if the rtc contains data that is not a valid deep sleep setting. If it  [... cut]
closed 2017-11-03 by marcelstoer