nodemcu/nodemcu-firmware #2157: Memory leakage

bastos1505, 2017-11-01:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
Every time  one module is loaded or when the http server receive a GET/POST the collectgarbage don't release all memory allocated. Even setting all variable to nil and after that, call the collectgarbage function.

### Expected behavior
After allocate a module/vars, use them, set them to nil and call a collectgarbage is expected to have the same heap size that had before the use of these modules/vars.

### Actual behavior
when some loaded module and/or vars to nil and call the function collectgarbage, the heap size don't return to original size.
After 110 seconds, the LUA machine execute a complete collectgarbage and it takes 10 seconds to release all memory allocated (module Lua.c, function dojob). I putted a counter in the function dojob (in lua.c file) and every time the counter reachs 500 I execute a complete collectgarbage. It's releasing memory much better. But still didn't solver the problem.

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
You can use this code bellow

**print(node.heap())
local mymod = require "mymodule"
local myvar = 0
....... --use the module functions
....... -- use myvar
......
mymod, package.loaded.mymodule = nil,nil
myvar = nil
collectgarbage("collect")
print(node.heap())**

or  this one

**server=net.createServer(net.TCP, 30)
    server:listen(80,function(conn)
        conn:on("receive", function(client,request)
            print(node.heap())
            local _, _, method, path, vars = string.find(request, "([A-Z]+) (.+)?(.+) HTTP")        
            if (method == nil)then
                _, _, method, path = string.find(request, "([A-Z]+) (.+) HTTP");
            end
            
            local _GET = {}
            local translateMgr = require "stdio"  -- my module            
            if (vars ~= nil)then
                for k, v in string.gmatch(vars, vars_pattern) do
                    _GET[translateMgr.getTranslate(k)] = translateMgr.getTranslate(v)
                end
            end
            k, v = nil, nil
            package.loaded.stdio, translateMgr = nil,nil
            method, path, vars = nil, nil, nil
            _GET,package.loaded.stdio, translateMgr = nil, nil, nil
            local http_var[#http_var + 1] = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n"
            http_var[#http_var + 1] = "{\r\n\"count\" : "..c1..",\r\n\"time\" : "..t1..",\r\n\"total\" : "..c2..",\r\n\"elapsed\" : "..t2.." \r\n}"

            local function httpsender(client)
                if( #http_var > 0 )then
                    client:send(table.remove(http_var,1))
                    collectgarbage("collect")
                else
                      client:close()
                      http_var = nil
                      client, request, conn = nil, nil, nil
                      collectgarbage("collect")
                      print(node.heap())
                 end
              end

              client:on("sent", httpsender)    
              httpsender(client)            
            print(node.heap()) 
        end)       
    end)**

```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.
Version 2.1.0/Master

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.
No hardware specific setup is necessary

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

Code1110, 2017-11-02:
The function `client:close()` caused memory leaks in my case. It's safe to remove that.

TerryE, 2017-11-02:
You are using conn in `httpsender` and hence this is a upval to on receive CB.  If you monitor the Lua registry function creation using `debug.getregistry()` then you will see that this is creating the resource growth that is eating heap.

-  The example also doesn't compile (`local http_var[#http_var + 1] = ...` is invalid syntax.
-  You've got lots of (uninitialised?) globals, e.g. `vars_pattern`, `k`, `v`, `c1`, `c2`, `t1`, `t2`.  `for` variables are automatically local within the scope of the do ... end and so  k, v = nil, nil` assigns globals to `nil`.
-  Incidentally you've got _lots_ of unneeded setting variables to nil, e.g. any locals are automatically destroyed as part of scope exit, so there is no point in setting them to nil before you exit.

So you could pretty much collapse your receive callback to the following, (a single send doesn't need buffering), though I would also question the value of reloading your `stdio``module on every request since it is used on every request.

Sorry but I don't think that this is a NodeMCU Lua issue.

closed 2017-11-02 by TerryE

TerryE, 2017-11-02:
> The function `client:close()` caused memory leaks in my case. It's safe to remove that.

No, but `conn:close()` would.

bastos1505, 2017-11-03:
Thank you Terry, for your reply. The code above, is part of my code. I omitted many valiable like c1,c2,t1,2...etc. But I realy used them.
The problem realy hapens, You can try to set a simple http server using server=net.createServer(net.TCP, 30)
    server:listen(80,function(conn) ....... 
and use your pc and make many GET. You'll see the heap memory decreasing at each GET. Or, you can create a module, load your module and destroy the module. You'll see that each time you load/destroy your heap will decrease.
Please, disconsider the code above. It's missing a lot of code. If you want I can send the complete code. No problem. It is just for eg. I didn't preocupate with the syntax :) . Sorry.

TerryE, 2017-11-03:
@bastos1505, Carlos,  the main point I was making was the first.  By referencing `conn` in  `httpsender` you are making it an upvalue to it and the on receive function, so you have a function in the registry which is cleared on the destructor for `conn`, but `conn` is an upvalue to the function and hence it won't get destroyed.  Yes, this leak is a bug, but a bug in your application.  Read my FAQ for further information.

bastos1505, 2017-11-03:
You are really fast TerryE :) Thank you again. 
Do you have some example of http server that have no memory leakage? A "fail safe code" :)