nodemcu/nodemcu-firmware #2160: wifi.sta.config breaks backwards compatibility

mikep01, 2017-11-02:
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature
the original wifi.sta.config("ssid","password") api

### Justification
needlessly breaks existing tools and user LUA code
it should be easy to check for a string vs table parm in the c impl to handle accordingly

### Workarounds
I can use the new API in my own code, but tools like the LuaLoader.exe are broken and perhaps unlikely to be fixed.

8<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2017-11-02:
The old API was deprecated in early spring I believe and was removed a few months later. I don't see why we should revert that.

As for LuaLoader there's https://github.com/GeoNomad/LuaLoader/issues/5.

mikep01, 2017-11-02:
“I don't see why we should revert that.”

I do understand it’s sometimes necessary to break backwards compatibility.  In this case I could see no justification for breaking it as it would be easy to preserve.

So my response would be: “Preserving backwards compatibility demonstrates professionalism and sets a good example to non-professional contributors”.  I mean no offense with “non-professional”, by that I mean people who don’t do development for living.  When developing for a living you learn very quickly that customers do not appreciate the expense of being forced to fix what was previously working.

Unfortunately I don’t see a response to your issue against GeoNomad so I don’t hold out much hope there.

But of course it’s not my place to say or complain as I’m not a real customer so I’ll change the bring up procedure for my users to remove the need for the LuaLoader.

From: Marcel Stör 
Sent: Thursday, November 02, 2017 4:57 PM
To: nodemcu/nodemcu-firmware 
Cc: mikep01 ; Author 
Subject: Re: [nodemcu/nodemcu-firmware] wifi.sta.config breaks backwards compatibility (#2160)

The old API was deprecated in early spring I believe and was removed a few months later. I don't see why we should revert that.

As for LuaLoader there's GeoNomad/LuaLoader#5.

mikep01, 2017-11-03:
I did this, seems to work:
1) Copy wifi_station_config function from the 1.5 version to the 2.1.0 version, renaming it to old_wifi_station_config.  Add a comment about it being for backwards compatibility.
2) Change:
  if(lua_istable(L, 1))
to:
  if(lua_isstring(L, 1))
	return old_wifi_station_config(L); 
  else if(lua_istable(L, 1))

No need to add it to the docs since as it's deprecated, nor any need to update/maintain the old function.
Now old code like the LuaLoader still works, new exploiters will use the new version.

marcelstoer, 2017-11-03:
> I don’t see a response to your issue against GeoNomad

That horse really seems dead, no commits in two years and no answers to issues.

> nor any need to update/maintain the old function.

It's more an academic question but why not? Every piece of code ultimately needs to be maintained.

@dnc40085 if you or other maintainers think we should re-add support for the old syntax then by all means go ahead...

MacSass, 2017-11-12:
Hi, I would appreciate some backward compatibility as well - mainly for the tool side. I agree there would be no need to support it officially, but if it can be done with no big hassle and does not have any negative side effects - I would appreciate it.
Unfortunately LuaLoader does not seem to be maintened, I also tried to contact the developer to see if he was willing to fix it, but did not even get a reply.
Unfortunately I also do not have any know-how in the language it was coded ... so I think we can´t hope for a fix on his side ...
Regards - Macsass

marcelstoer, 2017-11-12:
We can argue about this from different perspectives but when it comes to LuaLoader...

> I also tried to contact the developer to see if he was willing to fix it, but did not even get a reply.
> ...
> so I think we can´t hope for a fix on his side

Then why should a firmware that is still maintained care about supporting unmaintained tools? Isn't this a hint that maybe you should look for "better" tools (better could e.g. mean _under active development_)? Even a 10-months old overview like https://frightanic.com/iot/tools-ides-nodemcu/ from yours truly shows no less than 8 alternative uploaders.

Would be happy to hear from other maintainers as well.

MacSass, 2017-11-12:
Hi ,
I do agree that maybe there are more important things to think about and that it does not make sense to try riding a dead horse - unfortunately the feature set of LuaLoader is pretty unique. I checked out the other options but especially things like GPIO setting etc. is not available in any other tool yet (I´ll get with the programmer of ESPCut to see if he is willing to add what is missing from LuaLoader right now.

I leave it up to you guys to figure out what makes sense - I can live with whatever you decide. 

Thanks for the time you invest in this project!
Macsass

marcelstoer, 2017-11-26:
Closing as no other maintainer felt this be important enough to engage in.

closed 2017-11-26 by marcelstoer