nodemcu/nodemcu-firmware #2161: udp:send returns undocumented error

mikep01, 2017-11-02:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
no error (it's UDP so ignore it)
### Actual behavior
"PANIC: unprotected error in call to Lua API (file:line routing problem)"
The last log entry I had was a report that wifi was disconnected, I don't know whether it was still was or if it was just reconnected
### Test code
I cannot easily recreate the condition.  I would expect it's best found by a code inspection.  I imagine it's could be the port or IP are invalid on the receive, though it could be the chip is not in a state to send due to the wifi state.
    rcvSvr = net.createUDPSocket()
    rcvSvr:on("receive",
      function(cn,data, port, ip)
          cn:send(port,ip,"my response")
      end)
    rcvSvr:listen(MppPort)
    rcvSvr:ttl(32)
### NodeMCU version
2.1.0 master
### Hardware
nodemcu ESP8266 (reflashed)
8<------------------------ END BUG REPORT -------------------------------------

mikep01, 2017-11-02:
I see this in net.c line 555:
    err_t err = udp_bind(ud->udp_pcb, &laddr, 0);
    if (err != ERR_OK) {
      udp_remove(ud->udp_pcb);
      ud->udp_pcb = NULL;
      return lwip_lua_checkerr(L, err);
    }
where lwip_lua_checkerr(L,err) does a "case ERR_RTE: return luaL_error(L, "routing problem");"
The documentation reports the only result as nil.
I suppose easy to fix in the docs.  Not so much for us users as we would need to pcall all net functions?
Another topic that might need investigation is why the "receive" function would be called if the send is not ready to succeed?

mikep01, 2017-11-02:
I've pcall'd the send and found by reconnecting the wifi that there's nothing wrong with the target ip and port.  Looks like a setup timing problem - the udp send fails despite the receiving of a valid incoming UDP message.
If this wasn't lua I would add pcalls everywhere, but memory is tight...

cwrseck, 2019-05-31:
Unfortunately, as far as I can tell both:
```
    pcall(udpskt:send(UDPPort, bcast, str))
```
and
```
    pcall(udpskt.send(udpskt, UDPPort, bcast, str))
```
fail with:
PANIC: unprotected error in call to Lua API (demo.lua:line: bad argument #1 to 'pcall' (value expected))

The error seems to occur irregularly, every three or five hundred calls, and while the send()
failure isn't a problem, the PANIC is.  Is there in fact a way of adding pcall() to udpskt:send()?

C W Rose

nwf, 2019-06-01:
That's not how `pcall` works.  Try `pcall(udpskt.send, udpskt, UDPPort, bcast, str)`.

nwf, 2019-06-02:
Sorry, that's apparently not a 5.1-ism either; try `pcall(function() udpskt:send(UDPPort,bcast,str) end)`

closed 2020-05-27 by nwf