nodemcu/nodemcu-firmware #2164: Add API to set WiFi RF power

marcelstoer, 2017-11-03:
### Missing feature
Add API to set WiFi RF power

### Justification
When power supply is not a problem you might want to crank up the volume on the WiFi to reach that AP. It's available in Arduino: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L285
There's `	system_phy_set_max_tpw` in the SDK: http://espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf#page=31 -> chapter 3.1.13

@dnc40085 whadayathink?

### Workarounds
Move the device closer to the AP...

volatile666, 2017-11-08:
push
Shouldnt this be implementable easily?

paulblondeau, 2017-11-08:
I could really use this feature too.

marcelstoer, 2017-11-08:
> Shouldnt this be implementable easily?

Tell us once you tried 😜

volatile666, 2017-11-09:
I did. I failed (not having worked on node mcu before) :D

PR #2171 (2017-11-19): esp8266: WiFi method to adjust maximum TX power
Fixes #2164 

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

I added the requested method to set maximum TX power.  I wasn't sure if the API should be expressed in dBm or these quarter-dBm units that the espressif SDK uses, so I went w [... cut]
closed 2017-11-19 by marcelstoer